fventuri/gr-sdrplay3 #3: FindLibSDRplay.cmake registry path (win32)

guruofquality, 2021-01-16:
https://github.com/fventuri/gr-sdrplay3/blob/master/cmake/Modules/FindLibSDRplay.cmake also needs the path fixed

guruofquality, 2021-01-16:
Other typos and missing link stuff

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 260c1e3..dc224f2 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -70,6 +70,25 @@ ELSE()
     message(WARNING "C standard could not be set because compiler is not GNU, Clang or MSVC.")
 ENDIF()

+IF(CMAKE_CXX_COMPILER_ID STREQUAL "MSVC")
+    #fix min/max errors in gnuradio headers
+    add_definitions(-DNOMINMAX)
+ENDIF()
+
+########################################################################
+# Find boost
+########################################################################
+find_package(Boost "1.65" REQUIRED filesystem)
+
+if(NOT Boost_FOUND)
+    message(FATAL_ERROR "Boost not found")
+endif()
+
+if (MSVC)
+    add_definitions(-DBOOST_ALL_NO_LIB)
+    add_definitions(-DNOMINMAX)
+endif()
+
 ########################################################################
 # Install directories
 ########################################################################
diff --git a/cmake/Modules/FindLibSDRplay.cmake b/cmake/Modules/FindLibSDRplay.cmake
index 553c376..2739514 100644
--- a/cmake/Modules/FindLibSDRplay.cmake
+++ b/cmake/Modules/FindLibSDRplay.cmake
@@ -2,7 +2,7 @@ if(NOT LIBSDRPLAY_FOUND)
     # pkg_check_modules (LIBSDRPLAY_PKG libsdrplay)

     IF(WIN32)
-        GET_FILENAME_COMPONENT(SDRPLAY_API_DIR "[HKEY_LOCAL_MACHINE\\SOFTWARE\\SDRplaySDR\\API;Install_Dir]" ABSOLUTE)
+        GET_FILENAME_COMPONENT(SDRPLAY_API_DIR "[HKEY_LOCAL_MACHINE\\SOFTWARE\\SDRplay\\Service\\API;Install_Dir]" ABSOLUTE)

         if( CMAKE_SIZEOF_VOID_P EQUAL 8 )
             MESSAGE( "64 bit compiler detected" )
@@ -19,7 +19,7 @@ if(NOT LIBSDRPLAY_FOUND)
                 "${SDRPLAY_API_DIR}/inc"
                 )

-        find_library(LIBSDRPLAY_LIBRARIES NAMES sdrply_api.lib
+        find_library(LIBSDRPLAY_LIBRARIES NAMES sdrplay_api.lib
                 PATHS
                 "${SDRPLAY_API_DIR}/${EX_PLATFORM_NAME}"
                 )
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index ae66560..af43085 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -39,10 +39,13 @@ add_library(gnuradio-sdrplay3 SHARED ${sdrplay3_sources})
 target_link_libraries(gnuradio-sdrplay3
     gnuradio::gnuradio-runtime
     ${LIBSDRPLAY_LIBRARIES}
+    ${Boost_LIBRARIES}
   )
 target_include_directories(gnuradio-sdrplay3
     PUBLIC $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/../include>
     PUBLIC $<INSTALL_INTERFACE:include>
+    PRIVATE ${LIBSDRPLAY_INCLUDE_DIRS}
+    PRIVATE ${Boost_INCLUDE_DIRS}
   )
 set_target_properties(gnuradio-sdrplay3 PROPERTIES DEFINE_SYMBOL "gnuradio_sdrplay3_EXPORTS")

```

fventuri, 2021-01-17:
@guruofquality - Thanks for the fixes, Josh!

I just pushed all of your changes to master; let me know if I missed anything.

Franco

guruofquality, 2021-01-17:
Builds now, thanks!

closed 2021-01-17 by guruofquality