fventuri/gr-sdrplay3 #4: No module named 'sdrplay3' error

juriguen, 2021-05-27:
Hi!

I'm trying to make the SDRplay RSPdx module work in GNU Radio but, after installing the latest app version for Ubuntu and following the instructions of the readme file to build the block, I try to run a very simple example and get the error: ModuleNotFoundError: [image]

Any ideas what might I be doing wrong?

Thanks!

fventuri, 2021-05-27:
@juriguen - I am not really sure what went wrong, but we can try to troubleshoot the problem together with these few steps:

- To make sure the install part had no problems, please run again the command `sudo make install`in the directory where you built the sdrplay3 GNU Radio module; the output should contain lines like these (any warning or error here could be a sign of a problem):
```
$ sudo make insttall
...
-- Up-to-date: /usr/local/lib64/libgnuradio-sdrplay3.so
...
-- Up-to-date: /usr/local/lib64/python3.9/site-packages/sdrplay3/__init__.py
-- Up-to-date: /usr/local/lib64/python3.9/site-packages/sdrplay3/__init__.pyc
-- Up-to-date: /usr/local/lib64/python3.9/site-packages/sdrplay3/__init__.pyo
-- Installing: /usr/local/lib64/python3.9/site-packages/sdrplay3/sdrplay3_python.cpython-39-x86_64-linux-gnu.so
...
```

- Next, I'd like to make sure that the files for the Python module are there using `ls -l <sdrplay3 python module directory>` (the exact directory name on Ubuntu might be different from mine, since I am running Fedora, but you should see the correct directory name from the output of the first command):
```
$ ls -l /usr/local/lib64/python3.9/site-packages/sdrplay3/
total 268
-rw-r--r--. 1 root root    745 Feb 28 15:09 __init__.py
-rw-r--r--. 1 root root    640 Feb 28 16:05 __init__.pyc
-rw-r--r--. 1 root root    640 Feb 28 16:05 __init__.pyo
-rwxr-xr-x. 1 root root 258176 May 27 07:22 sdrplay3_python.cpython-39-x86_64-linux-gnu.so
```

- Finally, I'd like to make sure that the sdrplay3 Python module is able to find all its dependent shared libraries with the `ldd` command as follows (again the exact path here might be different from mine; see the directory name from the output of the first command):
```
$ ldd /usr/local/lib64/python3.9/site-packages/sdrplay3/sdrplay3_python.cpython-39-x86_64-linux-gnu.so
...
```
the output from the command above should list twenty or so shared libraries - none of them should have the words `(not found)` next to it.

I also remember that on some systems it is recommended to run the command `sudo ldconfig` after the `sudo make install` to make sure that the runtime linker/loader finds all the dependent libraries when it tries to load the module; you may want to run it there to see if it helps (and if it does, please let me know, and I will add a note about it in the README file).

Franco

juriguen, 2021-05-28:
Hi! Thanks for the answer

When I check the contents of the `sdrplay3` directory, I get twice as many files as you:
```
ls -l /usr/local/lib/python3/dist-packages/sdrplay3/
total 532
-rw-r--r-- 1 root root    745 may 27 09:22 __init__.py
-rw-r--r-- 1 root root    644 may 27 14:12 __init__.pyc
-rw-r--r-- 1 root root    644 may 27 14:12 __init__.pyo
-rw-r--r-- 1 root root 262408 may 27 11:38 sdrplay3_python.cpython-38-x86_64-linux-gnu.so
-rw-r--r-- 1 root root 262408 may 27 14:34 sdrplay3_python.cpython-39-x86_64-linux-gnu.so
```

I guess this is due to me trying to correct the issue yesterday by installing a newer version of Python, which I have removed today.

Then, the `ldd` command gives:
```
dd /usr/local/lib/python3/dist-packages/sdrplay3/sdrplay3_python.cpython-38-x86_64-linux-gnu.so 
	linux-vdso.so.1 (0x00007ffe5b09c000)
	libgnuradio-sdrplay3.so.1.0.0git => /usr/local/lib/x86_64-linux-gnu/libgnuradio-sdrplay3.so.1.0.0git (0x00007fddb4cc2000)
	libgnuradio-runtime.so.3.10.0git => /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.0git (0x00007fddb4b8d000)
	liblog4cpp.so.5 => /lib/x86_64-linux-gnu/liblog4cpp.so.5 (0x00007fddb4b49000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fddb4968000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fddb494d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fddb475b000)
	libsdrplay_api.so.3.07 => /usr/local/lib/libsdrplay_api.so.3.07 (0x00007fddb454a000)
	libgnuradio-pmt.so.3.10.0git => /lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.10.0git (0x00007fddb44df000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fddb44bc000)
	libvolk.so.2.4 => /lib/x86_64-linux-gnu/libvolk.so.2.4 (0x00007fddb41d0000)
	libboost_program_options.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0 (0x00007fddb4141000)
	libboost_thread.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007fddb4115000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fddb4108000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fddb4102000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007fddb407e000)
	libpython3.8.so.1.0 => /lib/x86_64-linux-gnu/libpython3.8.so.1.0 (0x00007fddb3b28000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fddb39d9000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fddb4d68000)
	liborc-0.4.so.0 => /lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007fddb3954000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fddb3926000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fddb390a000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fddb3905000)
```

The `sudo ldconfig` didn't solve the issue either... Any other ideas?

Toni

juriguen, 2021-05-28:
Hi again

The issue seemed to be that the `PYTHONPATH` and/or `LD_LIBRARY_PATH` variables had to include the `sdrplay3` directory, as they have pointed out to me in `https://chat.gnuradio.org/#/room/#gnuradio:gnuradio.org`

Now I get another error:
```
gr::log :ERROR: rspdx0 - SDRplay device not found: **********
Traceback (most recent call last):
  File "/home/toni/Downloads/gr-sdrplay3/examples/rspdx.py", line 208, in <module>
    main()
  File "/home/toni/Downloads/gr-sdrplay3/examples/rspdx.py", line 186, in main
    tb = top_block_cls()
  File "/home/toni/Downloads/gr-sdrplay3/examples/rspdx.py", line 84, in __init__
    self.sdrplay3_rspdx_0 = sdrplay3.rspdx(
RuntimeError: sdrplay device not found or invalid mode/antenna selection
```
Do you know how should I proceed?

Toni

juriguen, 2021-05-28:
Well, sorry I am sending so many messages, but I think I solved it. The example had an RSP selector field as "****" and I just erased that I left it blank. It now seems to work.

Thanks!
Toni

fventuri, 2021-05-28:
@juriguen - I am glad to hear that you were able to figure out and solve the problem!

If you have multiple RSP devices connected, you can use to RSP selector field to enter the serial number of the specific one you want to stream from.

Franco

closed 2021-05-28 by fventuri