fventuri/gr-sdrplay3 #5: Scan spectrum

juriguen, 2021-05-28:
Hi again

This might not be an issue of your software, but I have implemented a simple flow to try to change the center frequency of the RSPdx device in real time and it crashes (after some time running, during which I am unsure as to whether it actually tunes to different frequencies).

Do you know if your implementation of the device allows for the center frequency to be changed in real time?

Thanks!
[image]

fventuri, 2021-05-28:
@juriguen - I think changing the center frequency should work.
In the worst case scenario it should print some sort of error message, but definitely not crash.

Can you attach here the .grc file you used for your GNU Radio flowgraph so I can give it a try, and see if it works for me (and perhaps fix the problem if it doesn't)?

Thanks,
Franco

juriguen, 2021-06-23:
Hi again

I made the flow run, and changing frequencies sometimes worked, but some other times it did not. I will post the error another day, since I am working from home today and do not have access to the SDR device.

I do have one more concern, though. Does your module introduce any distortion when tuning a certain frequency band (that you know of)? I have scanned the RF spectrum a couple of times, then stored IQ samples in a file via a sink module. Then, I plot the entire spectrum using Matlab, but I do not get anything that makes sense. Instead, the result looks like a narrow band is amplified around the center frequency at each re-tune of the sweep. Not sure if this makes sense.

Thanks
Toni

fventuri, 2021-06-23:
@juriguen - Toni, thanks for the feedback

For the first problem, regarding the updates to the frequency not taking place all the times, I received a note by @SDRplay a week ago or so about something very similar happening with the SoapySDR module SoapySDRPlay.

He mentioned that the SDRplay API for updates to the sample rate, frequency, and gain (reduction) are asynchronous, i.e. a call to the `sdrplay_api_Update()` function to say update the frequency returns immediately, while the frequency on the RSP might be changed only a few ms later.
What he suggested (which is what he does in his RSPTCPServer) is to wait until the RX callback sets a non zero value for the paramater `rfChanged` (in the case of the frequency).
A few days ago I implemented his suggestions in the SoapySDRPlay module and it seems to have resolved the problem in his use case.

Since I suspect you are experiencing the same problem here with the `gr-sdrplay3` GNU Radio OOT module, I just created a new branch called `wait_for_updates` (https://github.com/fventuri/gr-sdrplay3/tree/wait_for_updates), that contains the same code changes (i.e. `set_center_freq()` won't return until the frequency on the RSP has actually changed).

Please build the `gr-sdrplay3` GNU Radio OOT module using the code from this branch and let me know if it addresses the first problem.

As per the second problem about the 'distortion', I am not sure I understand it exactly.
If you are running a flowchart in GNU Radio Companion, do you mind posting it (or at least the relevant parts) so I can try to reproduce the problem here to better understand it?
Also your comment about the 'narrow band' you are seeing makes me wonder what values you are using for the sample rate and for the bandwidth in your code - perhaps they are too small.

Franco

juriguen, 2021-06-24:
Hi again

The distortion I mean is that, given a sampling rate of 2MHz, for instance, when I scan the spectrum, store the IQ samples and then plot the entire spectrum with Matlab, I get like a seesaw effect in which about 500KHz seem to be further enhanced compared to the BW (which I have assummed is the same as the sampling rate).

I attach a screenshot of the GNUradio flow. If you want the grc file, let me know and I send it too. I also attach some pics of the spectrum as plotted in Matlab, where you will see the weird effect I was talking about.

If you think all this is a bit too much for a github issue, we may continue via email. Mine is juriguen@gmail.com

Thanks

[image]

[image]

fventuri, 2021-06-24:
Thanks @juriguen for the screenshots.

I don't remember how the block parameter `Bandwidth` set to `AUTO` works (I would have to look at the source code when I have some time), but to rule that out (or not), I would like you to run a couple more scans with the parameter `Bandwidth` in the RSPdx block set explicitly to 1.536MHz first and then to 5MHz to see if there's any difference in the Matlab spectrum.

Also did the new code in the branch `wait_for_updates` help with the problem of the frequency changes being skipped every so often?

I am OK with continuing this conversation here, because I could use this issue as a reference in the future if someone has a similar problem; if you would like to email me directly, my email address is in the git commit log (you can see it with the command `git log`).

Franco