fventuri/gr-sdrplay3 #6: Problem installing SDRPlay API for Jetson Nano (ARM Ubuntu)

Pratik310commits, 2021-06-23:
I have installed SDRPlay API 3.07, updated by GNURadio to latest, and followed the workflow given for Installation, still now able to successfully get SDRPlay working in conjunction with GNURadio Companion, for RSP1A.

fventuri, 2021-06-23:
@Pratik310commits - thanks for reporting this problem with the 'gr-sdrplay3' GNU Radio OOT module.

As mentioned in the README (https://github.com/fventuri/gr-sdrplay3/blob/master/README.md#important-notice), this GNU Radio OOT module requires GNU Radio version 3.9 (or newer); the screenshot you attached makes me think the Jetson Nano is  instead running on older version (3.7.13.4).
Another indication that 'gr-sdrplay3' might not be the GNU Radio OOT module you are running there is the fact that on the right hand side of your screenshot I see that the module group is called 'SDRplay' (instead of 'sdrplay3' - see screenshot below for what it should look like).

[image]

I also took a quick look at the non-Windows workflows on SDRplay's website and all the ones I found for GNU Radio seem to use the old version of SDRplay API (2.13 or something like that); could you send me the link (or a screenshot) to the workflow you used for installing the SDRplay API and GNU Radio on the Jetson Nano with ARM Ubuntu?

Franco

Pratik310commits, 2021-06-23:
Thanks for the quick response Franco.
The screenshot I am posted, is on the raspberry pi
On the Jetson nano, I couldn't all the steps mentioned in the workflow done.
I used the workflow as in the screenshot I'm posting

[image]

Also, how do I get the sample console app running?
I am unable to get it to run on my windows laptop and my raspberry pi

1. On Windows machine, it just cannot find the "sdrplay_api.h" file.
   
2. On the Raspberry Pi, I am getting errors like "Too few arguments to the function"
Is there any specific git repository I should clone on my raspberry pi local?
I was able to install the SDRPlay API and get it working with the GNURadio Companion, but I also need a console app, but unsuccessful attempts so far.

Could you help me sort these two issues out asap?

fventuri, 2021-06-24:
@Pratik310commits - thanks for posting the screenshot of the workflow - now I understand much better what you are trying to do.

I see the instructions in the workflow use the SDRplay API 2.13 and the github repository: https://gitlab.com/HB9FXQ/gr-sdrplay - to try to avoid any compatibility problem I would recommend that you follow what SDRplay instructions say, i.e. build the GNU Radio module 'gr-sdrplay' using that version on https://gitlab.com/HB9FXQ/gr-sdrplay and not this one.

Regarding your specific questions, I am not really able to help you much with Windows, since I use Linux here.

For the Raspberry Pi, since Raspberry OS is a derivative of Debian like Ubuntu, I would suggest that you first remove the files you installed from the git repo https://github.com/fventuri/gr-sdrplay3, reboot it, and then try again to run the commands in SDRplay workflow that you posted above.

If you get again an error like "Too few arguments to the function", please let me know what specific command you ran, and also please copy here a few lines before and after that error message to provide some useful context.

Franco

Pratik310commits, 2021-06-24:
"_build the GNU Radio module 'gr-sdrplay' using that version on https://gitlab.com/HB9FXQ/gr-sdrplay and not this one._"
I am a bit confused about this. Could you please explain again? Do I use it or not?
And for Raspberry Pi also, do I use 
"_remove the files you installed from the git repo https://github.com/fventuri/gr-sdrplay3, reboot it, and then try again to run the commands in SDRplay workflow that you posted above._"
I remove the ones for SDRPlay V3? and use the one given in the workflow? An older version?

I am sorry, I am very confused (not just a bit, but very)

fventuri, 2021-06-24:
@Pratik310commits - I really apologize that my comment made you more confused instead of helping.

There are currently two different versions of the SDRplay API that you can use with the RSPs and GNU Radio:

- SDRplay API version 2.X (I think the current version is 2.13) - this is the one that is used in the SDRplay workflow that you show above, and this is the one I suggest you try first because it is well established and should work without any problem there. The three components of this option are:
  - SDRplay API version 2.13 (see the SDRplay download link above and do make sure you download and install version 2.13)
  - the gr-sdrplay module from https://gitlab.com/HB9FXQ/gr-sdrplay (as per the instructions in the SDRplay workflow)
  - GNU Radio version 3.7.<something> (which probably comes standard with your Linux distribution)

- SDRplay API version 3.X (I think the current version is 3.07) - this is the newer version and for GNU Radio is still experimental (see below). The three components of this option are:
  - SDRplay API version 3.07 (see the SDRplay download link above and do make sure you download and install version 3.07)
  - the gr-sdrplay3 module from https://github.com/fventuri/gr-sdrplay3
  - GNU Radio version 3.9 or 3.10 (you will probably have to compile this from source and it may take a while; that's why I call this option still 'experimental')

The reason for my note above about removing the files you might have installed from https://github.com/fventuri/gr-sdrplay3 was because I was afraid you had different components from different versions installed on your Raspberry Pi there, and perhaps they would cause conflicts with each other.

I hope this explanation helps you solve your problems,
Franco

Pratik310commits, 2021-06-24:
Thanks alot
I'll look into this right away.

Pratik310commits, 2021-06-24:
On the Jetson Nano, I did the following things
1. Installed SDRPlay API 3.07 using the .run file with the description "API 3.07 (for 64bit OS)"
2. Installed GNURadio 3.9.0 using _https://wiki.gnuradio.org/index.php/InstallingGR_
3. Followed the workflow with the following commands
    a. sudo apt get install swig git cmake libboost all dev
    b. mkdir ~/Dev && cd ~/Dev
    c. git clone https://github.com/fventuri/gr-sdrplay3.git
    d. cd gr-sdrplay3 && mkdir build && cd build
    e. cmake .. && make && sudo make install && sudo ldconfig

At 3.e., i had the following issue

[image]

Do you recommend I try the SDRplay API version 2.13 along with GNU Radio version 3.7 ?

fventuri, 2021-06-24:
@Pratik310commits - congratulations! I think you are almost there.

It looks like you just need to install the 'pybind11-dev' package - please run this command:

> sudo apt-get install pybind11-dev

and after that, please try running step 3.e again

Franco

Pratik310commits, 2021-06-24:
I did actually the missing package
The I had the issue that i previously raised - errors in the libraries installed

/home/pratik/Pictures/error_after_pybind.png

fventuri, 2021-06-24:
@Pratik310commits

I looked at the first of those errors:

> 'overload_cast_impl' in namespace 'pybind11::detail' does not name a template type

and, according to some pages I found with Google, it seems it might be due to a missing include.
Since that template function is defined in the header `pybind11/detail/common.h`, could you edit the source file `python/bindings/rsp1_python.cc`, and add this line after the include's at the beginning of that file (i.e. after line 12)?

> #include <pybind11/detail/common.h>

after saving this change, please try running the command `make` again.

Also what exact version of Ubuntu are you running there?
If I have time tonight, I'll try to build a VM with the same version of Ubuntu (I am running Fedora Core 34 here, that's why I don't experience the same problems) to see if I can reproduce the same errors here.

Franco

Pratik310commits, 2021-06-24:
I will try modifying the header file and post an update. Also, I am using Ubuntu 18.04, on ARM

fventuri, 2021-06-24:
@Pratik310commits

How did you install GNU Radio 3.9 on your Ubuntu 18.04 using the instructions here: https://wiki.gnuradio.org/index.php/InstallingGR ?

I just tried here two different ways:
> sudo add-apt-repository ppa:gnuradio/gnuradio-releases
> sudo apt-get update
> sudo apt install gnuradio

but these commands installed GNU Radio 3.8 (not 3.9, despite of they say in that installation document); I checked it with this command:

> sudo apt list --installed | grep -i gnuradio

and I see that the version is 3.8

I also tried with these commands:

> sudo add-apt-repository ppa:gnuradio/gnuradio-master
> sudo apt-get update
> sudo apt install gnuradio

but the last command (apt install gnuradio) failed with a several dependency errors.

By any chance did you install GNU Radio from source, using 'git clone  https://github.com/gnuradio/gnuradio.git', and all the other commands listed there?

Franco

Pratik310commits, 2021-06-25:
I did the same
First installed with 
`sudo add-apt-repository ppa:gnuradio/gnuradio-releases
sudo apt-get update
sudo apt install gnuradio`

It gave me error saying, 3.8 was found, not 3.9
I checked the version using 
`gnuradio-config-info --version`
It said 3.8

So then I did 
`sudo add-apt-repository ppa:gnuradio/gnuradio-master
sudo apt-get update
sudo apt install gnuradio`

This installed without errors.
I did not use the 'git clone https://github.com/gnuradio/gnuradio.git'

fventuri, 2021-06-25:
@Pratik310commits - thanks for your reply.

I did some research on the dependency error I saw earlier when trying the repository `ppa:gnuradio/gnuradio-master`, and it turns out I am having the exact same problem discussed in this thread on the `discuss-gnuradio` mailing list: https://lists.gnu.org/archive/html/discuss-gnuradio/2021-02/msg00129.html

According to the reply there it is not possible to install GNU Radio 3.9 on Ubuntu 18.04 because of the problem with the library  `libvolk`, so I was wondering if you could check which version of the libvolk package you have on your system and which Ubuntu repository does it come from.

Franco

Pratik310commits, 2021-06-25:
@fventuri 
Thanks for these quick responses and researches/
I will let you asap..

Pratik310commits, 2021-06-25:
@fventuri 
I am new to Linux and Ubuntu and struggling, could you help me with the command or method to find out the version and repository of the libvolk package?

Pratik310commits, 2021-06-25:
Used the command `sudo apt list --installed | grep -i libvolk`

Result:

libvolk2-bin/bionic,now 2.3.0-2~ubuntu18.04.1~ppa1 arm64 [installed,automatic]
libvolk2-dev/bionic,now 2.3.0-2~ubuntu18.04.1~ppa1 arm64 [installed,automatic]
libvolk2.0/now 2.0.0-2~bionic1 arm64 [installed,local]
libvolk2.2/now 2.2.1-2~ubuntu18.04.1~ppa1 arm64 [installed,local]
libvolk2.3/bionic,now 2.3.0-2~ubuntu18.04.1~ppa1 arm64 [installed,automatic]

fventuri, 2021-06-25:
@Pratik310commits - thanks for the versions of the `libvolk` library

This morning I was finally able to install the package `libvolk2.2` here, run all the other steps in the instructions (instead of `swig` I used `pybind11`, since this is the new type of Python bindings being used by GNU Radio starting with version 3.9), and the build was successful for me:

[image]

The only difference I see is that my Ubuntu 18.4 runs on the amd64 (Intel) architecture because that's what I have on my PC, while yours is an ARM Ubuntu, but I thought they were very similar.

I understand you are new to Linux, but at this point since those messages are C/C++ errors, perhaps you could ask someone with C/C++ experience to look into them, and see if that include suggestion I mentioned earlier (https://github.com/fventuri/gr-sdrplay3/issues/6#issuecomment-867931232) could perhaps help.

The other option could be to try to run the workflow that uses the old SDRplay API version 2.13.

Franco

Pratik310commits, 2021-06-25:
Yes, I will try the API version 2.13 first
Then once that starts working, I will spend time with the C/C++ issues

Pratik310commits, 2021-06-28:
I installed SDRPlay API 2.13 and GNURadio 3.7
I got the following error during the last step pf the workflow: `cmake .. && make && sudo make install && sudo ldconfig`

[image]

Then I tried installing GNURadio 3.8 and got the error:

[image]

So basically, since SDRPlay 3.07 and GNURadio 3.9 weren't installed correctly, I tried but couldn't successfully get to install SDRPlasy API 2.13 along with GNURadio 3.7
I was hopefuly about SDRPlay 2.13 along with GNURadio 3.7

Pratik310commits, 2021-06-28:
Hi @fventuri 
Apart from the above update, something much weirder happened
Since I couldn'tget anything to install correctly, I just thought of installing SDRPlay 3.07 and GNURadio 3.9 AGAIN, after trying to install SDRPlasy API 2.13 along with GNURadio 3.7

and  it worked
I dont which residue files helped or what happerned,orif they updated something on SDRPlay's end 

But look at this:

[image]

[image]

fventuri, 2021-06-28:
@Pratik310commits - Congratulations!

I am very happy to see if finally worked with the new SDRplay API 3.07!

(BTW the earlier error about the 'wrong format' typically happens when for some reason x86/x64 Intel binaries or objects get mixed together with ARM binaries/objects).

Franco

Pratik310commits, 2021-06-28:
@fventuri 
Oh yea, the binaries not being for the appropriate architecture, but I was wondering why did it start behaving that way all of a sudden.
So I gave up on that and just tried one last go at the latest APIs.
Thanks for the quick responses a lot.

Pratik310commits, 2021-07-13:
Hi @fventuri 
I might be knocking at the wrong door here, but I want to know whether there is an updated version of the sample_app.c code.
Because the one that I am trying to work with (after a long time after successfully installing SDRPlay), gives errors like "Too few arguments"
This is definitely because of changes made to the API itself, so I looked for an updated version but had no luck
I am trying to read the code and also debug to find out if I can make the necessary changes myself but again no luck.
any suggestions?

Pratik310commits, 2021-07-13:
*Update*
I found the latest API sample_example_app.c in the API documentation pdf. the code still has issues in getting compiled.
I am not new to C/C++ programming, but I am still not able to get it to work.
It shows undefined reference to all the functionas

[image]

fventuri, 2021-07-14:
@Pratik310commits 
Those `undefined reference` error messages normally happen when you don't add the sdrplay_api library to the load/link command.

From your screenshot above I can't see the exact command you are using, but assuming you have something like `gcc ...... sdrplay_api_example.c`, you probably need to append something like `-L/usr/local/lib -lsdrplay_api`:

```
gcc ...... sdrplay_api_example.c -L/usr/local/lib -lsdrplay_api
```

Hope this helps,
Franco

Pratik310commits, 2021-08-05:
Hi @fventuri 
Came back to working on jetson nano after working on raspberry pi for couple of weeks, to try and replicate the work on jetson nano

The simple flowgraph won't execute 
[image]
Error: File "/home/pratik/EMSA/GRC_Projects/sweep_no_headblock/FFT_Sink/FFT_Sink.py", line 34, in <module>
    import sdrplay3
ModuleNotFoundError: No module named 'sdrplay3'

And I tried using C++ as the output language, it says "samp_rate" variable is not defined, although I have no such variable in the flowgraph

Generating: '/home/pratik/EMSA/GRC_Projects/sweep_no_headblock/FFT_Sink/FFT_Sink'
Generate Error: (NameError("'samp_rate' is not defined",), ['set_samp_rate(${samp_rate});', 'set_center_freq(${center_freq});', 'set_bandwidth(${bandwidth});', 'set_gain_mode(${AGC});', 'set_gain(-${if_gRdB}, "IF");', 'set_gain(-${rf_gRdB}, "RF");', 'set_freq_corr(${freq_corr});', 'set_dc_offset_mode(${dc_offset_mode});', 'set_iq_balance_mode(${iq_balance_mode});', 'set_agc_setpoint(${agc_set_point});', 'set_rf_notch_filter(${rf_notch_filter});', 'set_dab_notch_filter(${dab_notch_filter});', 'set_biasT(${biasT});', 'set_debug_mode(${debug_mode});', 'set_sample_sequence_gaps_check(${sample_sequence_gaps_check});', 'set_show_gain_changes(${show_gain_changes});'])

[image]

Pratik310commits, 2021-08-05:
I added the samp_erate variable back in, still doesn't work.

however, when I created a console app, it works perfectly fine

[image]

fventuri, 2021-08-05:
@Pratik310commits 
- for the first issue with the message `No module named 'sdrplay3'`, you may want to see if that module is in your Python search path `sys.path` (https://docs.python.org/3/library/sys.html#sys.path); you can find out what `sys.path` is set to by adding the following debug line in the Python script right before the line with `import sdrplay3`:
```
print(sys.path)
```
If the module `sdrplay3` is not under any of the directories listed with that statement, you may want to reinstall it by going to the directory where you built it, and running again the command:
```
sudo make install
```
which should also print the path where that module is being installed - make sure that that path is indeed in your Python `sys.path`

- as per the second question about the `samp_rate` variable, please attach the `.grc` file you are running, so I can see if I can reproduce the problem here in order to understand where the problem is.

Franco