fventuri/gr-sdrplay3 #7: Isssues installing gr-osmocom with SDRPlay 3 and GNURadio 3.9

Pratik310commits, 2021-07-26:
Hi @fventuri 
I want to connect a HackRF One device with my NVIDIA Jetson nano. I have GNURadio 3.9 and SDRPay API 3.07 currently installed.
I used "https://github.com/osmocom/gr-osmosdr" and followed the instructions but got numerous errors
Do you suggest a  different source/repo?
I am attaching the text file with the errors.
[osmocom_errors.txt](https://github.com/fventuri/gr-sdrplay3/files/6879355/osmocom_errors.txt)

fventuri, 2021-07-26:
@Pratik310commits 
looking at the error message you sent me and at the `gr-osmosdr` web page here: https://osmocom.org/projects/gr-osmosdr/wiki/GrOsmoSDR, I am pretty sure you are missing some dependencies.

If you notice in that page, it says:

> Before building the block you have to make sure that all the dependencies (see list of supported devices above) you are intend to work with are properly installed. The build system of gr-osmosdr will recognize them and enable specific source/sink components thereafter.

For instance in your case you'll need the `libhackrf` library (I think in Ubuntu it is in the package `libhackrf-dev`), `libsndfile`, and perhaps others.
Also to make your life easier, I would suggest you disable support for devices you are not going to be using, like the 'FUNcube Dongle', by adding the string `-DENABLE_FCD=0` (for the 'FUNcube Dongle') to the `cmake` command.

Franco

Alster, 2021-09-14:
@Pratik310commits looks like [this patch](https://lists.gnu.org/archive/html/discuss-gnuradio/2020-09/msg00055.html) can help you