fventuri/gr-sdrplay3 #8: RSPDuo: startup frequencies

Alster, 2021-09-14:
Hi!

Just noticed that at flowgraph startup RSPDuo in DualTuner & IndependentRX mode both channels uses same frequency, even if they are different. Looks like channel 2 always use frequency from channel 1. When you change frequency later, in realtime, all works fine.

In this animation, we see that at the beginning frequencies are different (8e6 and 18e6), but signals looks the same. Also we see that after changing frequency from 8e6 to 9e6 and then return back to 8e6 - signals now different, all works fine
[image]

Here is rspduo options:
[image]

And example flowgraph:
[rspduo_test.grc.zip](https://github.com/fventuri/gr-sdrplay3/files/7161100/rspduo_test.grc.zip)

fventuri, 2021-09-14:
@Alster - thanks for the reporting the problem.

I only had a few minutes this morning before work to take a quick look at it.

I added this debug line right before the call to`'sdrplay_api_Init()` in the source file `lib/rsp_impl.cc` (https://github.com/fventuri/gr-sdrplay3/blob/master/lib/rsp_impl.cc#L524):

```
std::cerr << "just before sdrplay_api_Init() - freq0=" << device_params->rxChannelA->tunerParams.rfFreq.rfHz << " - freq1=" << device_params->rxChannelB->tunerParams.rfFreq.rfHz << std::endl;
```

I rebuilt and reinstalled the `gr-sdrplay3` module, and then ran your flowgraph with the module with the debug statement, and on my computer it does display two different center frequencies:
```
just before sdrplay_api_Init() - freq0=1.8e+07 - freq1=8e+06
```

If you have some time, do you mind running the same quick test and let me know what you see there?
If you see two different center frequencies too, then I'll take a deeper look at your example tonight after work.

Franco

Alster, 2021-09-14:
Yes, see the same output:
`just before sdrplay_api_Init() - freq0=1.8e+07 - freq1=8e+06`

I just added another debug line, right [before this one](https://github.com/fventuri/gr-sdrplay3/blob/7c54c3f432890ea1758ff13776602f02be48bc16/lib/rspduo_impl.cc#L97), to see previous frequency value before setting new:
```cpp
std::cerr
    << "Freq " << tuner + 1 << ":"
    << " from " << get_independent_rx_channel_params(tuner)->tunerParams.rfFreq.rfHz
    << " to " << freq
    << std::endl;
```

And got this output:
```bash
Freq 1: from 2e+08 to 1.8e+07 # ok
Freq 2: from 2e+08 to 8e+06 # ok
# looks like SDRplay uses 2e+08 as default frequency for all channels, and this is all ok
just before sdrplay_api_Init() - freq0=1.8e+07 - freq1=8e+06 # our previous debug-line, which tells us that all is ok
# now I`m changing frequencies for both channels to see what happens
Freq 1: from 1.8e+07 to 1.9e+07 # add 1e6
Freq 1: from 1.9e+07 to 1.8e+07 # minus 1e6
# all is ok, but for second channel we see that previous value is 1.8e+07, but not 8e+06
Freq 2: from 1.8e+07 to 9e+06 # add 1e6
Freq 2: from 9e+06 to 8e+06 # minus 1e6
```

Strange, but I faced with similar problem[ when making own gnu radio block using your SoapySDRPlay3 module](https://github.com/pothosware/SoapySDRPlay3/issues/45). I was thinking that it is my bug, but now looks like this bug can come from sdrplay api.

Hope it helps you!

fventuri, 2021-09-15:
@Alster - I ran a few more tests tonight, and I think you have a point about a possible bug in the SDRplay API.

What I did was to add the following debug statement right after the call to `sdrplay_api_Init()` (i.e. before this line: https://github.com/fventuri/gr-sdrplay3/blob/master/lib/rsp_impl.cc#L529):
```
std::cerr << "just after sdrplay_api_Init() - freq0=" << device_params->rxChannelA->tunerParams.rfFreq.rfHz << " - freq1=" << device_params->rxChannelB->tunerParams.rfFreq.rfHz << std::endl;
```
and this is what I saw running your flowgraph:
```
just before sdrplay_api_Init() - freq0=1.8e+07 - freq1=8e+06
just after sdrplay_api_Init() - freq0=1.8e+07 - freq1=1.8e+07
```
in other words, it looks like `sdrplay_api_Init()` resets the center frequency of the second tuner to the same value of the first tuner (18MHz in your example).

To work around this problem with `sdrplay_api_Init()` I created a new branch called `fix_tuner2_frequency_for_independent_rx` that you can find here: https://github.com/fventuri/gr-sdrplay3/tree/fix_tuner2_frequency_for_independent_rx - the only code change in this branch is a call to `set_center_freq` immediately after `sdrplay_api_Init()` to set back the center frequency of the second tuner to the correct value in the specific case of the RSPduo running in dual tuner mode with independent receivers.

Please build and run the code from the `fix_tuner2_frequency_for_independent_rx`, and let me know if it addresses your problem; if it does, I'll run some more tests tomorrow night to make sure it does not trigger any unexpected problems, and then I'll merge it into the `master` branch.

Also I am not sure at this point if `sdrplay_api_Init()` resets other settings of the second tuner to the same values of the first tuner (besides the center frequency); if you find other problems like this one, please let me know so I can take care of those settings as well in the new code.

Franco

Alster, 2021-09-15:
Yes, your hack from `fix_tuner2_frequency_for_independent_rx` branch successfully solved issue, and all works fine!
What about another settings: looks like Gain AGC bugged too
Anyway, this issue can be closed for me, thank you for your work!