fventuri/gr-sdrplay3 #9: How do I change frequency while the flowgraph is running?

Pratik310commits, 2021-10-04:
Hi @fventuri 
I was (and still am) using an older version of the GNURadio API on my Raspberry Pi.
I need to change the frequency of the radio after a certain interval of time and I was doing right after tb.Start(True).
I did not want to use threads because I also want to get binary dumps in a different file each time I change the frequency, using the file_sink block.
I made this work in python by 
1. Locking the flowgraph
2. Changing the frequency
3. Disconnecting the file sink block
4. Changing the output filename
5. Re-connecting the file sink block
6. Unlocking the flowgraph
7. Let it run for a certain amount of time and then repeat the above steps

I am doing this in a loop
Now I have started working on a Ubuntu Machine that has the latest API version and thus the blocks and the python code generated are also changed a lot.
The logic I was using previously isn't working anymore. I see a lot of things in the main() function - qa.exec(), tb.show(), signals, etc.
That is not an issue, my problem is - I am unable to change the frequency like I was able to, using the above logic

Could you please
First - suggest a better way, if there is, to do the above steps 
second - help me solve this problem of not being able to change the frequency and save it on a different file each time

Pratik310commits, 2021-10-04:
The following is the code that works on the Raspberry Pi

`tb.set_rsp1a_freq_mhz(0)
    tb.Start(True)
    
    print("Waiting for 2s to let the device settle")
    time.sleep(2)
    start = datetime.datetime.now()
    print("Starting @: ", start.strftime("%b_%d_%Y_%I_%M_%S"))
    freq = 1
    while(freq <= 20):
        ts = datetime.datetime.now().strftime("%b_%d_%Y_%I_%M_%S")
        
        # partial_filename = dirName + fileNamePrefix + ts + "_"
        tb.lock()
        print("Change freq to " + str(freq))
        tb.set_rsp1a_freq_mhz(freq)
        time.sleep(1)
        # print("Changing file name")
        tb.disconnect(tb.blocks_file_sink_0)
        final_filename = dirName + fileNamePrefix + "_rf" + str(freq) + "M_"+ ts + ".bin"
        # final_filename = dirName + fileNamePrefix + "_rf" + str(freq) + "M_.bin"
        tb.blocks_file_sink_0 = blocks.file_sink(gr.sizeof_gr_complex*4096, final_filename, False)
        tb.connect((tb.fft_vxx_0, 0), (tb.blocks_file_sink_0, 0))
        
        tb.unlock()
        print('Frequency set to ' + str(tb.get_rsp1a_freq_mhz()))
        # print("Wait for 1")
        time.sleep(sweepTime)
        freq = freq + 1
    
    tb.stop()`

Pratik310commits, 2021-10-04:
I am unable to get this code to work with the latest API
It just stops after changing the frequency the first time.
It just stops printing anything and the files created is 0 bytes.
I have to kill the process each time

Pratik310commits, 2021-10-04:
I just added the line after reconnecting the file_sink block and it works now (although still not till the end)
I added
`blocks_file_sink_0.set_unbuffered(False)`
Now my First question is still the same
Do you have suggestions for changing frequency and file name every time?

fventuri, 2021-10-04:
Pratik,
I just wrote the attached GNU Radio Python script that captures the I/Q samples for each MHz from 88MHz to 108MHz for 10 seconds; I ran it and it worked without any errors here (I am running the latest version of GNU Radio on Linux Fedora).

Franco

[change_frequency.py.gz](https://github.com/fventuri/gr-sdrplay3/files/7278279/change_frequency.py.gz)

AugustoMarziani, 2021-10-05:
Hi everyone! I think I have some similar issues. I am reading frequencies from an external file and I created an output to send to a scope and then change the frequency of the SDR in dual channel same frequency mode.
Each time the frequency it's changed I get the following error:
gr::log :ERROR: rspduo0 - sdrplay_api_Update(Tuner_Frf) Error: sdrplay_api_RfUpdateError
For a while the frequency switch from one to the other with ghosting effects and then the error stop and the frequency it's right.
Is there a way to avoid it? I am using API v3 and the branched version of gr-sdrplay3. 
Any help will be really appreciated.
Thanks in advance!

fventuri, 2021-10-05:
@Pratik310commits @AugustoMarziani - thanks for your reports about this bug.

I was able to reproduce that behavior last night with the simple flowgraph attached to this note.
I stopped the python script using the command `kill -ABRT <pid>`, which created a core dump; then with the command `coredumpctl debug` (or `gdb` if your Linux distribution doesn't have `coredumpctl`), I was able to see what functions the threads were running when I interrupted that python script that was frozen.

The `gdb` command I ran to see the output below was:
```
thread apply all bt
```

And these are the important pieces from that output:
```
Thread 3 (Thread 0x7ff0b3fff640 (LWP 3731)):
#0  0x00007ff0f8464a8a in __futex_abstimed_wait_common64 () from /lib64/libpthread.so.0
#1  0x00007ff0f845e2c0 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#2  0x00007ff0e9885b70 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib64/libstdc++.so.6
#3  0x00007ff0c210990b in gr::sdrplay3::rsp_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/local/lib64/libgnuradio-sdrplay3.so.1.0.0git
#4  0x00007ff0e9ed5827 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#5  0x00007ff0e9e8ea55 in gr::block_executor::run_one_iteration() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#6  0x00007ff0e9ee0014 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#7  0x00007ff0e9ed4465 in gr::tpb_container::operator()() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#8  0x00007ff0e9ed46a4 in gr::thread::thread_body_wrapper<gr::tpb_container>::operator()() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#9  0x00007ff0ea2070b2 in thread_proxy () from /lib64/libboost_thread.so.1.75.0
#10 0x00007ff0f8458299 in start_thread () from /lib64/libpthread.so.0
#11 0x00007ff0f8570353 in clone () from /lib64/libc.so.6

Thread 2 (Thread 0x7ff0c1de7640 (LWP 3668)):
#0  0x00007ff0f8464a8a in __futex_abstimed_wait_common64 () from /lib64/libpthread.so.0
#1  0x00007ff0f845e2c0 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#2  0x00007ff0e9885b70 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib64/libstdc++.so.6
#3  0x00007ff0c210b8d3 in gr::sdrplay3::rsp_impl::stream_callback(short*, short*, sdrplay_api_StreamCbParamsT*, unsigned int, unsigned int, int) () from /usr/local/lib64/libgnuradio-sdrplay3.so.1.0.0git
#4  0x00007ff0c210cb06 in gr::sdrplay3::rsp_impl::stream_A_callback(short*, short*, sdrplay_api_StreamCbParamsT*, unsigned int, unsigned int, void*) () from /usr/local/lib64/libgnuradio-sdrplay3.so.1.0.0git
#5  0x00007ff0c1eece1c in strmHandlerThread(void*) () from /usr/local/lib64/libsdrplay_api.so.3.07
#6  0x00007ff0f8458299 in start_thread () from /lib64/libpthread.so.0
#7  0x00007ff0f8570353 in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7ff0f82fa740 (LWP 3664)):
#0  0x00007ff0f8464a8a in __futex_abstimed_wait_common64 () from /lib64/libpthread.so.0
#1  0x00007ff0f845e2c0 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#2  0x00007ff0ea206273 in boost::condition_variable::wait(boost::unique_lock<boost::mutex>&) () from /lib64/libboost_thread.so.1.75.0
#3  0x00007ff0ea206497 in boost::thread::join_noexcept() () from /lib64/libboost_thread.so.1.75.0
#4  0x00007ff0e9eebe21 in gr::thread::thread_group::join_all() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#5  0x00007ff0e9edd5a0 in gr::top_block_impl::restart() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#6  0x00007ff0e9edd798 in gr::top_block_impl::unlock() () from /usr/local/lib64/libgnuradio-runtime.so.3.10.0git
#7  0x00007ff0c2a3fafd in ?? () from /usr/local/lib64/python3.9/site-packages/gnuradio/gr/gr_python.cpython-39-x86_64-linux-gnu.so
#8  0x00007ff0c2a4036a in ?? () from /usr/local/lib64/python3.9/site-packages/gnuradio/gr/gr_python.cpython-39-x86_64-linux-gnu.so
#9  0x00007ff0c29e1613 in ?? () from /usr/local/lib64/python3.9/site-packages/gnuradio/gr/gr_python.cpython-39-x86_64-linux-gnu.so
#10 0x00007ff0f875f8f2 in cfunction_call () from /lib64/libpython3.9.so.1.0
#11 0x00007ff0f8751677 in _PyObject_MakeTpCall () from /lib64/libpython3.9.so.1.0
#12 0x00007ff0f874e14e in _PyEval_EvalFrameDefault () from /lib64/libpython3.9.so.1.0
#13 0x00007ff0f8755fe3 in function_code_fastcall () from /lib64/libpython3.9.so.1.0
#14 0x00007ff0f875e3a1 in method_vectorcall () from /lib64/libpython3.9.so.1.0
#15 0x00007ff0f874e05e in _PyEval_EvalFrameDefault () from /lib64/libpython3.9.so.1.0
#16 0x00007ff0f8755fe3 in function_code_fastcall () from /lib64/libpython3.9.so.1.0
#17 0x00007ff0f874938c in _PyEval_EvalFrameDefault () from /lib64/libpython3.9.so.1.0
#18 0x00007ff0f874800d in _PyEval_EvalCode () from /lib64/libpython3.9.so.1.0
#19 0x00007ff0f87c4635 in _PyEval_EvalCodeWithName () from /lib64/libpython3.9.so.1.0
```

A few important things to observe from this output:
- the function that freezes is `tb->unlock()` in thread 1; the reason why it hangs is because it is running a 'thread::join_all()` waiting for all the flowgraph running threads to join
[... cut ...]
- the problem is that there seems to be a race condition between thread 2 (the sdrplay3 `stream_callback`) and thread 3 (the GUN Radio `work` function), which are both waiting for a mutex, and therefore never completing.

Now that I have some idea of what is going on (please repeat the same steps when your scripts hang, to see if it is for the same reason in your cases), I'll spend some more time tonight after work to see exactly what is causing it and how to fix it.

I'll send you an update once I have more information.

Franco

[simple_test_unlock.py.gz](https://github.com/fventuri/gr-sdrplay3/files/7285647/simple_test_unlock.py.gz)

AugustoMarziani, 2021-10-05:
Thanks a lot! I will try to replicate what you did to see if it gives the same error.
I have to open another issue to ask for another problem I can not understand. I am working on a university project, if you can help me solving these problems, maybe we can also share the final product!

Pratik310commits, 2021-10-05:
Oh my! Thanks a lot for that detailed debug information.
I will check whether mine is caused by the same reasons and post it.
(Also working on a university project hahaha, I can relate with you @AugustoMarziani )

AugustoMarziani, 2021-10-05:
@Pratik310commits we can get in touch if you want. We can share information also by email.
marzianiaugusto@gmail.com

fventuri, 2021-10-06:
@Pratik310commits @AugustoMarziani

I suspect the race condition we are seeing is caused by me changing the value of the ring buffers head and tail variables inside the `start()` function without acquiring the mutex locks and without notifying the condition variables.

I added them and I just pushed the code to the new branch `fix_race_condition_on_ring_buffer`; I am running the simple Python script from yesterday and it hasn't blocked so far.

Please use this new branch to rebuild the `gr-sdrplay3` GNU Radio module, and let me know if it fixes the problem for you too; if it does, I'll merge this change into the master branch.

Also, now that I understand better what `tb.unlock()` does (not sure exactly what `tb.lock()` does, to be honest), I think that the current approach might have a problem, since the center frequency of the RSP gets changed _before_ invoking `tb.unlock()`, which does restart the flowgraph; in other words I suspect at the end of each file you might have a few samples that were captured at the new center frequency.

Perhaps a better way to do what you are trying to do is by having the frequency sweep loop use `tb.stop()` and `tb.start()` directly, i.e. something like this (I haven't tried to run it, but it should give you a good idea of what I mean):
```
    for center_freq in ...:
        print('I/Q capture frequency:', center_freq, 'MHz')
        tb.blocks_file_sink_0.open(f'/tmp/iq_capture_{center_freq}.bin')
        tb.stop()
        tb.wait()
        tb.sdrplay3_rspdx_0.set_center_freq(center_freq * 1e6)
        tb.blocks_file_sink_0.do_update()
        tb.start()
        time.sleep(5)
``` 

Finally I am excited to hear you are both working on some new university project; years ago I used to be a professor of Electronics (at the University of Parma, Italy), so I am very interested in your academic activities.

Franco

Pratik310commits, 2021-10-06:
Just yesterday I thought about using start() and stop() instead of lock-unlock
I was about to try it today.
I'll try both and post an update.

And I'm glad you are showing so much interest in our projects
Thanks alot @fventuri

AugustoMarziani, 2021-10-06:
@Pratik310commits can you please explain in detail what you did? I am not so practical about python programming. I am actually using the gr-companion GUI to put together blocks.

Pratik310commits, 2021-10-06:
I think I spoke too soon with the tb.start-tb.stop approach working perfectly
Even that misbehaves
`Changing freq to 9
gr::log :ERROR: rsp1a0 - sdrplay_api_Init() Error: sdrplay_api_AlreadyInitialised
Frequency set to 9
Changing freq to 10
gr::log :ERROR: rsp1a0 - sdrplay_api_Update(Tuner_Frf) Error: sdrplay_api_RfUpdateError
gr::log :ERROR: rsp1a0 - sdrplay_api_Init() Error: sdrplay_api_AlreadyInitialised
Frequency set to 10`

This happens some times, and some time it works fine

Pratik310commits, 2021-10-06:
@AugustoMarziani 
I added the blocks I needed - stream to vector -> FFT -> File sink
And in  the generated Python file, I am making changes to change the centre frequency dynamically at run time and save the sink in different files each time.
i am doing this in the main() function

Pratik310commits, 2021-10-06:
@fventuri 
on calling `set_center_freq` , sometimes it gives me `Error: sdrplay_api_NotInitialised` and some times `Error: sdrplay_api_AlreadyInitialised`
This is when I am using the tb.start() and tb.stop() approach
Even when combined the two - tb.stop() + tb.lock() and tb.unlock() + tb.start(), it didn't work

Since tb.lock-unlock wasn't giving me these Initialisation errors, I thought I'll combine the two and see what happens...

Pratik310commits, 2021-10-06:
@fventuri 
I did the following things (to sum it up) and I am getting the following issues

1. Tried the normal way of tb.stop - tb.start()
Issue : it gives me `Error: sdrplay_api_NotInitialised` and some times `Error: sdrplay_api_AlreadyInitialised`

2. Tried the tb.lock -  tb.unlock using the separate branch 'fix_race_condition_on_ring_buffer' (git clone -b fix_race_condition_on_ring_buffer https://github.com/fventuri/gr-sdrplay3.git )
Issue: Still misbehaves the same way. in fact, couple of times it gave me the desired output, rest of the times it got blocked and I had to kill the process.
(The zip that I attached above "sweep_change_freq.py" used the same file for the newer branch)

3. Tried combination of start-stop and lock-unlock, with the newer branch  'fix_race_condition_on_ring_buffer' as well as the master branch, still gave the initialisation errors

Pratik310commits, 2021-10-06:
Apparently, GNURadio 3.7 requires python 2.7 which isn't supported in Ubuntu 20.04, if I am not wrong
So I can't install a previous version GNURadio on Ubuntu 20.04
Are suggestions for how should I proceed?
@fventuri

Pratik310commits, 2021-10-06:
NVM
Found the sdrplay 2.13 installation file