fventuri/gr-sdrplay3 #10: Problem managing two data stream from SDRplay RSPduo

AugustoMarziani, 2021-10-05:
Hi everybody! 
I am working on a project. The aim of this is to measure the power of the two signals at the same frequency coming from the RSPduo (diversity mode). I need to filter the two channels to reduce the noise bandwidth before measuring the power. 
The problem is that, when I try to filter the data through the FFT low pass filter block, the processing crash.
If I use only one filter, and the other channel without, and setting the option of 4 threads, it goes on. If I use a filter on both channels it gets stuck.
I do not understand why it happens. It does not generate errors on the gr console, but I only see a horizontal line in the graphs and the code seems to be stuck (other operations are not working).

Attached the block diagram of the first part of the project.

The first block is a python custom block. It is a pass thorugh for the two signal, while it retrieve the frequency for the tuner from an external file.

Any suggestion to solve this problem? Can someone help me closing this project?
Thanks in advance!!

[image]

AugustoMarziani, 2021-10-06:
@fventuri thanks a lot. So now the original error seems to be solved. But another one appeared. It seems that in this way the SDR only work with the Automatic Gain Control setting turned on. If I disable it, I get the error:
gr::log :ERROR: rspduo0 - sdrplay_api_Init() Error: sdrplay_api_Fail
I tried it with the "fix_race_condition_on_ring_buffer"   branch and the simplified code. With the same configuration but single channel it works. Advanced options are: IF att =40, RF att =70, AGC set point = -30.

I just realized that in dual channel mode, the attenuation levels are not the same as for the single channel. There is a maximum level of RF att = 40 dB, more than this generate the above error. Is there a way to bypass that?

fventuri, 2021-10-07:
@AugustoMarziani - thanks for the debug info; it helps me understand what is the problem.

The LNA state is not a gain (or gain reduction), but it is rather a value/index that tells the RSPduo what the the gain reduction in the RF stage should be (and it is dependent on the RSP model, on the frequency range, and on the antenna input).

To better understand what I mean, please take a look at the tables in section 5 (page 38) of the SDRplary API Specification Guide (I think the latest version is this one: https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.07.pdf): as you can see for the RSPduo the LNAstate value can only be in the range [0-9] (depending on the frequency), and therefore the value of 40 is out of range.

Also another document that you might find useful is the 'RSPduo Technical Information' (https://www.sdrplay.com/wp-content/uploads/2018/06/RSPDuo-Technical-Information-R1P1.pdf); in that document at pages 14-16 you'll find a block diagram of the RSPduo that should help you have a good idea of how all the various components of the RSPduo work together.

Ciao,
Franco

fventuri, 2021-10-07:
@AugustoMarziani 
Ciao Augusto,
first of all, sorry for the confusion: I use the direct value of the LNA state in a different project (the SoapySDR driver SoapySDRPlay3), while in this GNU Radio module `gr-sdrplay3` the user selects both IF and RF attenuation in dB; I should have checked that before writing my previous comment.

Anyway, in the properties for the `gr-sdrplay3` block, under 'Gain Options', I see two attenuation settings:
- IF attenuation
- RF attenuation (which gets translated to LNA state inside the module code)

[image]

I just ran the RSPduo dual tuner example from the repository (https://github.com/fventuri/gr-sdrplay3/blob/master/examples/rspduo_dual_tuner.grc), changing the dual tuner configuration to 'diversity reception', and setting the IF attenuation to 40dB and the RF attenuation to 70dB (as shown in the picture above), and in my case the flowchart ran without any errors (I can see the RF spectrum changing in the frequency sink window).

Franco

AugustoMarziani, 2021-10-11:
Hi Franco,
during these days I was on leave, and I will not be available also during this week.. I will try to test again everything tomorrow. Can you please tell me which API and gnuradio version are you using?
Thanks in advance,
Augusto