fventuri/gr-sdrplay3 #15: Bandwidth setting AUTO does not "set the bandwidth to the largest value below the sample rate"

kirat68, 2021-11-05:
Hello Franco

Would be nice if you could check this issue. Setting Bandwidth to AUTO takes the lowest value of the allowed bandwidths and NOT "the largest value below the sample rate".

Test settings:
Sample rate = 5E6
Bandwidth = Auto

Result: Bandwidth = 2E5
Expected: Bandwidth = 4E6

Otherwise working very good. Thank you very much !!!

fventuri, 2021-11-06:
@kirat68 - thanks for pointing out this problem.

It should now be fixed in the `master` branch.

Please also note that 4e6 is not one of the bandwidth supported by SDRplay RSPs; you can find the list of supported values in the SDRplay API Specification guide at page 12: 200k, 300k, 600k, 1536k, 5M, 6M, 7M, and 8M.

Franco

kirat68, 2021-11-06:
Yes yes just realized that 4E6 was not the correct value and corrected my comment before seeing yours :-)

kirat68, 2021-11-06:
Expected value should be in this case 5E6. Specification should be:
"set the bandwidth to the largest value below **or equal* to the sample rate"
Am I right ?

fventuri, 2021-11-06:
@kirat68 - please note that in the same commit yesterday (https://github.com/fventuri/gr-sdrplay3/commit/c74162f756081b7d23fb4a229b73cbb086918c0b), I changed that wording to "compatible with".

Franco

kirat68, 2021-11-09:
Perfect !!!