fventuri/gr-sdrplay3 #16: Output stdout to GRC message port

kirat68, 2021-11-16:
This feature would be nice to have:
- Output of stdout to a gnuradio message port for further processing inside the python script

fventuri, 2021-11-17:
Tarik,
I looked at the source code for the `gr-sdrplay3` GNU Radio module and I don't see anywhere where it is writing directly to `stdout` (or to the C++ stream `std::cout`).

I also just ran the FM receiver example (https://github.com/fventuri/gr-sdrplay3/blob/master/examples/FM-Receiver-SDRplay.grc) for about 30 seconds, and the only messages I see on the console are warning and info log messages from the `GR_LOG_*` functions (or perhaps macros).

Are these log messages the "output to stdout" that you are referring to in your request?

Franco

kirat68, 2021-11-17:
Olá Franco,

Sorry was not enough concise. Yes I meant the log messages that are printed in the console.
Would be great the have them printed to a message port. Would be then available in the grc python code for further actions. Nice also would be to have the choice to route them to console (as it is now) or to message port or to both.

For example, catching "overload detected" could be nice to trigger appropriate actions ...

Tarik