fventuri/gr-sdrplay3 #17: Module sdrplay3 not found

gom-4, 2021-11-16:
Hi everyone,
I've already installed gr-sdrplay3 and the API without any problem. However, when trying to execute gr-sdrplay3/examples/FM-Receiver-SDRplay with an RSPduo I get a ModuleNotFoundError: No module named 'sdrplay3'.
I'm using Ubuntu20.04 & GNURadio 3.9.3.

Can anyone help me with this?

The gnuradio console response:

```
Generating: '/home/user/Dev/gr-sdrplay3/examples/top_block.py'

Executing: /usr/bin/python3 -u /home/user/Dev/gr-sdrplay3/examples/top_block.py

Traceback (most recent call last):
  File "/home/user/Dev/gr-sdrplay3/examples/top_block.py", line 38, in <module>
    import sdrplay3
ModuleNotFoundError: No module named 'sdrplay3'

```

fventuri, 2021-11-17:
The first thing I would try is just:
```
sudo ldconfig
```
since on some Linux distributions the `ld` cache needs to be refreshed when a new shared library is installed.

If that doesn't help, please try to run this simple Python script:
```
#!/usr/bin/env python3
import sys
print(sys.path)
```

it should print a list of directories (your Python 'path'); please go through them and make sure one of them contains a directory called `sdrplay3` (in my case, on Fedora 35, the directory is `/usr/local/lib64/python3.10/site-packages`, but yours on Ubuntu 20.04 will likely be different).

If you do find that `sdrplay3` directory in your Python path, please make sure it contains a shared library called `sdrplay3_python.cpython-310-x86_64-linux-gnu.so` (or something very similar, depending on the exact version of Python you are running there).

Finally, run the command `ldd -r` on that shared library; in my case here the full command would be:
```
ldd -r /usr/local/lib64/python3.10/site-packages/sdrplay3/sdrplay3_python.cpython-310-x86_64-linux-gnu.so
```
(but yours is going to be slightly different), and make sure it doesn't show any dependent library as 'not found' or other messages about 'undefined symbol'.

Hope this helps,
Franco

gom-4, 2021-12-02:
Solved, thank you!

Now,  I'm having another issue when trying to use my RSPDuo.

When I first connect it I can find it by typyng `lsusb` as:
`Bus 001 Device 008: ID 1df7:3020`

However, when executing a flowgraph in gnuradio, my devide suddenly gets disconected and I receive this error:

```
gr::log :ERROR: rspduo0 - SDRplay device not found: 
Traceback (most recent call last):
  File "/home/radio/Dev/gr-radio_astro/examples/DSPIRA/DSPIRA_Spectrometer.py", line 754, in <module>
    main()
  File "/home/radio/Dev/gr-radio_astro/examples/DSPIRA/DSPIRA_Spectrometer.py", line 732, in main
    tb = top_block_cls()
  File "/home/radio/Dev/gr-radio_astro/examples/DSPIRA/DSPIRA_Spectrometer.py", line 236, in __init__
    self.sdrplay3_rspduo_1 = sdrplay3.rspduo(
RuntimeError: sdrplay device not found or invalid mode/antenna selection
```
And after that, `lsusb` doesn't find the devide.

I would appreciate your help, thank you so much

fventuri, 2021-12-03:
@gom-4 This is very strange; I've seen a few cases where the service `sdrplay_api` crashes, but I've never seen a case where the USB device disappears altogether like you have.

I would like to see if there's any error message from the SDRplay API in the logs.
These are the steps to look for possible error messages:
- in the RSPduo block in GRC, under 'Properties -> Other Options', enable the property called `SDRplay API debug mode (DEBUG)`
- save and run the GRC flowgraph until it gives you that error
- run these commands to search for messages from the service `sdrplay_apiService` in the logs:
```
grep sdrplay_ /var/log/messages
```
```
journalctl -b | grep sdrplay_
```

I hope that the messages in those logs will give us a better idea of the reason for this problem.

Franco

gom-4, 2021-12-08:
Hi Franco,

The problem was caused by my virtual machine USB configuration. Finally managed to solve it changing the controller.

Thank you for your help and congratulations on your great work.

closed 2021-12-10 by gom-4