fventuri/gr-sdrplay3 #19: Bulk/Isochronous transfer mode

pabs77, 2021-11-29:
Hey Franco :), 
I'm trying to get my sdrplay rspduo to change the data transfer mode, which is part of the API. I found it quite difficult to find and had to go through a lot of trouble with the bindings and cmake builds on Windows. I think im close to achieve it .. But maybe it can be added as another instance or for the init parameters of the sdr such as: set antenna or bandwidht.. it is at the end just a 1 or 0 flag. Maybe I dindnt see something or maybe I'm totally wrong and there is a way to change the transfer mode. Please correct me if I'm wrong. 

Have a nice day and thanks for the good work you guys have been doing ! 
Sincerly, 
Pablo

fventuri, 2021-11-29:
Thanks for the nice words, @pabs77 !
Kudos on your work to get it to run under Windows!

I tried to compile GNU Radio under Windows 10 a few months ago, so I could try the `gr-sdrplay3` module with it, but in the end I had to give up, because I do not have much experience with Windows (I run Linux here most of the time), and I wasn't able to fix some of the errors while trying to build GNU Radio (I think it was version 3.9).

Anyhow, your request seems pretty easy to implement; I'll give it a try tonight after work, and I'll let you know how it goes.

Franco

fventuri, 2021-11-30:
@pabs77 
I just created the new branch called `add_bulk_transfer_mode` (https://github.com/fventuri/gr-sdrplay3/tree/add_bulk_transfer_mode), that has a new method `set_bulk_transfer_mode(<boolean>)` to allow to select the USB transfer mode (the default is isochronous, as per SDRplay API Specification document).

Please give it a try when you have time, and if it looks good to you, I'll merge those changes in the `master` branch.

Franco

pabs77, 2021-11-30:
Thx Franco this is awesome ! Im trying to build it but got lots of issues with pybind11 and it tells me it cannot find some files like rsp_pydoc.h and all of those _pydoc.h. I'll keep on trying and let you know whatever I achieve.

Have a nice day