fventuri/gr-sdrplay3 #20: RSPduo RF and IF gain

flocentblack, 2022-01-19:
First, thanks for creating this! Enjoyed some 80m CW this morning in stereo!

I understand that the RF and IF gain values are listed as attenuation instead of gain like on the SDRduo chart and on the chart, I believe the gain to be the combination of RF and IF gain.  I've read that the IF gain has a range of +20dB to +59dB,  How are is this expressed as attenuation, what are acceptable values for RF and IF gain in gr-sdrplay3?

Also, I tried using variables in the RF and IF attenuation fields, and got errors.  Maybe my  variables were set to invalid values?

fventuri, 2022-01-19:
@flocentblack - glad to hear you enjoyed some 80m CW for breakfast!

In this `gr-sdrplay3` module, the two gain parameters (`IF` and `RF`) are indeed gains, i.e. higher values mean higher signal gain (or less attenuation, if you prefer to look at it that way).

Since the SDRplay RSP family of devices 'thinks' in terms of 'gain reduction' (i.e. attenuation), the values you want to assign to the `IF` and `RF` gain are always negative: for instance the `IF` gain goes from -20 (highest signal gain) to -59 (lowest signal gain); similarly the gain values for the `RF` gain go from 0 (highest signal gain of the RF chain) to -N, where the value of N depends on the specific RSP device and frequency band, as per the 'Gain Reduction Tables' in the SDRplay API Specification guide ([https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.09.pdf](https://www.sdrplay.com/docs/SDRplay_API_Specification_v3.09.pdf)) on page 38.

Please note that all the gain values in the `set_gain()` methods are in dB.

Feel free to reach out if you have any other question.

73,
Franco

flocentblack, 2022-01-20:
Thanks Franco for clarifying the IF and RF gain values.

In regards to variables it appears that the block is passing the variable id to the API instead of the variable value:

Generate Error: **(ValueError("could not convert string to float: 'rf_gain_1'")**, ["set_sample_rate(${sample_rate if rspduo_mode == 'Single Tuner' else sample_rate_non_single_tuner})", "${'set_center_freq(' + center_freq + ')' if rspduo_mode.nindeprx == '1' else ''}", "${'set_center_freq(' + center_freq0 + ', 0)' if rspduo_mode.nindeprx == '2' else ''}", "${'set_center_freq(' + center_freq1 + ', 1)' if rspduo_mode.nindeprx == '2' else ''}", 'set_bandwidth(${bandwidth})', 'set_antenna("${antenna_both if rspduo_mode.nchan == \'2\' else antenna}")', "${'set_gain_mode(' + AGC + ')' if rspduo_mode.nindeprx == '1' else ''}", "${'set_gain_mode(' + AGC0 + ', 0)' if rspduo_mode.nindeprx == '2' else ''}", "${'set_gain_mode(' + AGC1 + ', 0)' if rspduo_mode.nindeprx == '2' else ''}", '${\'set_gain(\' + str((-float(if_gRdB))) + \', "IF")\' if rspduo_mode.nindeprx == \'1\' else \'\'}', '${\'set_gain(\' + str((-float(if_gRdB0))) + \', "IF", 0)\' if rspduo_mode.nindeprx == \'2\' else \'\'}', '${\'set_gain(\' + str((-float(if_gRdB1))) + \', "IF", 1)\' if rspduo_mode.nindeprx == \'2\' else \'\'}', '${\'set_gain(\' + str((-float(rf_gRdB))) + \', "RF")\' if rspduo_mode.nindeprx == \'1\' else \'\'}', '${\'set_gain(\' + str((-float(rf_gRdB0))) + \', "RF", 0)\' if rspduo_mode.nindeprx == \'2\' else \'\'}', '${\'set_gain(\' + str((-float(rf_gRdB1))) + \', "RF", 1)\' if rspduo_mode.nindeprx == \'2\' else \'\'}', 'set_freq_corr(${freq_corr})', 'set_dc_offset_mode(${dc_offset_mode})', 'set_iq_balance_mode(${iq_balance_mode})', 'set_agc_setpoint(${agc_set_point})', 'set_rf_notch_filter(${rf_notch_filter})', 'set_dab_notch_filter(${dab_notch_filter})', 'set_am_notch_filter(${am_notch_filter})', 'set_biasT(${biasT})', 'set_debug_mode(${debug_mode})', 'set_sample_sequence_gaps_check(${sample_sequence_gaps_check})', 'set_show_gain_changes(${show_gain_changes})'])
>>> Failure`

flocentblack, 2022-01-20:
Another thing that I observed this morning was that while in independent RX mode tuned to the same frequency, the RF gain for RX0 affects the RF gain for RX1.  This is particularly noticeable on my system being that RX0 is connected to a 80 meter inverted L and RX1 is connected to a Loop-on-the-Ground antenna without a preamp, which has a noise floor below the noise floor of my RSPduo. Setting RF attenuation so that RX0 is not overloading, signals from RX1 disappear into the noise.  Setting RF attenuation to 0 causes RX0 to be overloaded, but signals from RX1 are now above the noise floor.  I've confirmed this with multiple trials, listening to headphones with RX0 to the left channel and RX1 to the right channel.

With "Show IF gain changes" enabled, I'm seeing only 'gRdb and lnaGRdb, with no distinction between which tuners these gains are being applied to, and changes to RX1 attenuation have no effect on either of these values.

Thank you for any help you can offer!

fventuri, 2022-01-20:
@flocentblack - thanks for reporting both bugs.
I'll have sometime to look at them tonight after work (it's almost 9am here in Florida).

Regarding the first issue with 'rf_gain_1' being passed as a string instead of a float, I would like to be able to reproduce the problem here (because I suspect it might happen in more than one place) - would it be possible for you to attach the actual GNU Radio Companion .grc file you used (or a simplified version of it that causes the same error)?

Thanks,
Franco

fventuri, 2022-01-21:
@flocentblack - thanks for the .grc file.
I was able to reproduce the problem here with that file and I fixed the callback generation code in the RSPduo GRC block.

I just pushed the code to the `master` branch; please run `git clone` to get the latest version from the repository, install it, and try you GRC flowgraph again.

Franco

flocentblack, 2022-01-21:
Its working great! On to building a more capable receiver...

Thanks again and 73,

Eric / NQ8J

closed 2022-01-21 by flocentblack