fventuri/gr-sdrplay3 #21: Actual RSPduo center frequency is different than commanded frequency

flocentblack, 2022-01-23:
When starting a flow graph, the initial frequency and often when changing the center frequency within the same band, is correct.

When changing the RSPduo center frequency for a different band, or back to the initial band, there is an offset in the RSPduo's actual center frequency, along with this error:

"rspduo :error: sdrplay_api_Update(Tuner_Frf) Error: sdrplay_api_RfUpdateError"

Further attempts would cause the band to be stuck one one band, as well as the same error message.

Changing from 80 to 40 meter selections, I have gone back and forth without issue. Changing from 80 to 20 or 40 to 20, I have consistency replicated this issue.  I wonder if this has something to do with the RSPduo changing from its 2-12MHz band pass filter to its 12-30MHz band pass filter.

Setting the initial value for 20 meters, I can create the same error going to 40/80.

Varying the frequency using a transmitter into a dummy load, offsets of approximately 14KHz to 48KHz were observed. I have not been to determine what causes variation in offset.

Restarting the flow graph is successful in restoring the RSPduo center frequency back to it's initial value.

I have also observed that tuning the RSPduo like a direct conversion receiver causes stuttering and the same errors.  Don't know if it is related.  I have been using the RSPduo as an LO and then do the VFO tuning using the Frequency Xlating FIR filter.

Attached is a flow graph that can be used to replicate this problem.

[test6.txt](https://github.com/fventuri/gr-sdrplay3/files/7920742/test6.txt)

fventuri, 2022-01-23:
@flocentblack - Eric, thanks for reporting this problem.

I saw in the past a similar error message when the center frequency was changed too fast - see this comment: https://github.com/fventuri/gr-sdrplay3/issues/5#issuecomment-866848230 (and related thread).
I was wondering if this could be your case too.

Also I saw your flowchart uses the RSPduo in dual-tuner mode, which may introduce some complexities; I wonder if you could try to run a similar frequency change in single tuner mode (you can just send the RSP output to a null sink for this test) to see if it happens in that case too.

73,
Franco K4VZ

flocentblack, 2022-01-24:
Single tuner, dual tuner (diversity) and dual tuner (independent RX) work fine into a null sink(s). RX frequency entered as variable, toggling between 80 and 20 meter frequencies, I could not get any errors.

I eliminated blocks, one by one until the error went away.  I was down to one Frequency Xlating Filter into a null sink without any errors.  It looks like two of those blocks at the same is are causing some sort of disruption on my system.

Thanks Franco.  73,

Eric / NQ8J

fventuri, 2022-01-24:
@flocentblack - thanks for the detailed and very useful research on this issue.

Would you mind posting the simplest GNU Radio flowgraph that shows the problem (which I understand is the dual-tuner case sending their outputs to a pair of frequency xlating FIR filters), so I can run it here with my RSPduo to see if the issue occurs here too?

Thanks Eric,
Franco

flocentblack, 2022-01-24:
Here you go.  Thanks!

[test8.txt](https://github.com/fventuri/gr-sdrplay3/files/7925971/test8.txt)

fventuri, 2022-01-25:
@flocentblack - Eric, thanks for GNU Radio Companion flowgraph.

I took your flowgraph, generated the Python script with GRC, then I simplified that Python script to the attached Python script, where I was able to reproduce your problem with different pairs of central frequencies, in all cases the lower one was < 12MHz and the higher one was > 12MHz, confirming your hypothesis.

You can run the Python script as follows for instance:
```
./erictest.py 11.99e6 12.01e6
```

As I was going through editing the original script generated by GRC, I noticed the following two lines in the method `set_lo` of that script:
```
        self.sdrplay3_rspduo_0.set_center_freq(self.lo, 0)
        self.sdrplay3_rspduo_0.set_center_freq(self.lo, 1)
```
As you can see, this is exactly the case of the issue I mentioned yesterday, where the code invokes the `set_center_freq` two times, one immediately after the other.

To confirm that this is the case, I ran the very same example with the code from the branch `wait_for_updates` (https://github.com/fventuri/gr-sdrplay3/tree/wait_for_updates), which comes from the branch `master` plus the code changes to wait for the frequency change to be completed before returning from the update method (see here: https://github.com/fventuri/gr-sdrplay3/commit/470b63d04e716978b9bb82c5c28467998df1ab32).
Using the code from the `wait_for_updates` branch, I re-ran the Python script and this time I don't see that error message anymore.

As an aside, I also noticed that in the GRC flowgraph you sent me, the two center frequencies are always set to the same value (variable `lo`); in this case it might be better to use the so called 'Dual Tuner (diversity reception)' mode, since it keeps all the settings for the two channels exactly the same.

73,
Franco

[erictest.zip](https://github.com/fventuri/gr-sdrplay3/files/7930047/erictest.zip)

flocentblack, 2022-01-25:
Thanks Franco for investigating this.  I would like to use the diversity reception mode, but my antennas are wildly different in gain, one being a is  loop on the ground, which has -20 to -30dB gain and the other an 80 meter inverted L. I need to run the input to tuner 2 wide open and then adjust the rf gain for tuner 1 as needed.  When I was using SDRUno in diversity mode, I could unlink the RF gains and adjust them independently. If you would like to add independent rf gain adjustment, that would be great, otherwise it sounds like using the wait for updates branch will be the solution for me.  

Thank you very much for the time and effort you have spent on creating this module as well as troubleshooting my issues.

73,
Eric

fventuri, 2022-01-25:
@flocentblack - thanks for the suggestion Eric.
I used SDRUno only a few times (and I've never tried diversity reception in SDRUno), so I wasn't aware that it allows to control the gains independently.
I think it should be possible to do the same in the GNU Radio module; tonight after work I'll see if I can make that change and I'll let you know.

73,
Franco

flocentblack, 2022-01-25:
Speaking of gains, in my opinion, using LNA state instead of gain makes for easier adjustment with only 7/9 positions instead of adjusting the slider value in dB until the LNA state flips to the next state.  But I’m fine if you orefer it the way it is.  I’m just thrilled to be able to still copy stations when there is QSB!

Eric

fventuri, 2022-01-27:
Eric,
I implemented your suggestion of having independent gain controls for dual tuner diversity reception mode in the new branch `rspduo_dual_tuner_fixes` (https://github.com/fventuri/gr-sdrplay3/tree/rspduo_dual_tuner_fixes).
I also made some improvements to the RSPduo code adding methods that allow to change both center frequencies, or both (IF and RF) gains with just one method call (which then calls the SDRplay API function `sdrplay_api_Update()` only once); this to avoid the problem with waiting for one center frequency to be done updating before being allowed to update the other center frequency.

I like your idea of also being able to set the LNA state values directly, but I would like to work on that change after this one is fully completed, since the LNA state values change is orthogonal to this one and I'd rather move step by step (see also the note about GNU Radio 3.10 below).

When you have time, please try the code in the `rspduo_dual_tuner_fixes` branch to make sure it works as expected; after you confirm it looks good to you, I'll merge it into the `master` branch; I'll also port the same changes to the `gnuradio-3.10` branch, since I am currently maintaining two versions of the `gr-sdrplay3` module, one for GNU Radio 3.9+ (the one in the `master` branch), and one for GNU Radio 3.10+ (the one in the `gnuradio-3.10` branch).

73,
Franco

flocentblack, 2022-01-28:
Thanks Franco! I've uninstalled gr-sdrplay3-master and installed gr-sdrplay3-rspduo_dual_tuner_fixes. I haven't noticed the offset problem I had earlier after switching from <12MHz to >12MHz and back , but am still getting an API error.  Also (I may have misunderstood you), I'm not seeing the independent RF gain adjustment in diversity mode.

I have gone as far as trying to load the flowgraph after uninstalling and the gr-sdrplay3 block is red and says missing block, to verify that the block is not gr-sdrplay3-master.

flocentblack, 2022-01-29:
^^^ The above post has been edited for clarity. ***

fventuri, 2022-01-29:
Eric, first of all thanks for giving it a try.

Let's start with the issue of the two independent RF gain adjustments in diversity mode.

In the new branch `rspduo_dual_tuner_fixes` in the `examples` directory you should see a very simple test case GRC flowgraph called `rspduo_dual_tuner_diversity_reception` (https://github.com/fventuri/gr-sdrplay3/blob/rspduo_dual_tuner_fixes/examples/rspduo_dual_tuner_diversity_reception.grc).

Please open that simple example in GNU Radio Companion, click on 'Properties' for the RSPduo block and under the 'Gain Options' tab you should see the independent gain controls for 'RX0' (tuner A) and 'RX1' (tuner B); if you don't, please make sure that the file `sdrplay3_rspduo.block.yml` on your system (which is probably in the directory `/usr/local/share/gnuradio/grc/blocks`) matches the one from the new branch (https://github.com/fventuri/gr-sdrplay3/blob/rspduo_dual_tuner_fixes/grc/sdrplay3_rspduo.block.yml).

Franco

flocentblack, 2022-01-29:
I opened the example flowgraph and no independent RF gain controls. Also the correct file is in the `/usr/local/share/gnuradio/grc/blocks` directory, verified by size and comparing changes from the master.  Weird huh?

flocentblack, 2022-01-29:
When I uninstalled the master version, all I did was `make uninstall`, from the master build directory. is there another step I am missing?

fventuri, 2022-01-29:
Eric, I agree - it is very weird.
Are you running GNU Radio 3.9 or 3.10 ?

Also using the very simple test case from the `examples` directory, could you generate the Python script from GNU Radio Companion and attach it here, so I can compare it with mine, to see if there are any differences that might help understand what's going on?

Thanks,
Franco

flocentblack, 2022-01-29:
I'm running GNU Radio 3.10.

[rspduo_dual_tuner_diversity_reception.txt](https://github.com/fventuri/gr-sdrplay3/files/7963935/rspduo_dual_tuner_diversity_reception.txt)

fventuri, 2022-01-29:
Eric,
the Python file you sent me seems to confirm that somehow your GRC block yaml file is still the old one; for instance the first few lines of the RSPduo block in your file have the following:
```
        self.sdrplay3_rspduo_0.set_sample_rate(2000e3)
        self.sdrplay3_rspduo_0.set_center_freq(100e6)
        self.sdrplay3_rspduo_0.set_bandwidth(0)
        self.sdrplay3_rspduo_0.set_antenna("Both Tuners")
        self.sdrplay3_rspduo_0.set_gain_mode(False)
        self.sdrplay3_rspduo_0.set_gain(-40, 'IF')
        self.sdrplay3_rspduo_0.set_gain(-0, 'RF')
``` 

while here I have:
```
        self.sdrplay3_rspduo_0.set_sample_rate(2000e3)
        self.sdrplay3_rspduo_0.set_center_freq(100e6)
        self.sdrplay3_rspduo_0.set_bandwidth(0)
        self.sdrplay3_rspduo_0.set_antenna("Both Tuners")
        self.sdrplay3_rspduo_0.set_gain_modes(False, False)
        self.sdrplay3_rspduo_0.set_gain(-40, -40, 'IF')
        self.sdrplay3_rspduo_0.set_gain(-0, -10, 'RF')
```

Notice how the AGC (gain mode) and `set_gain()` calls are different.

I just ran a quick search on the GNU Radio discussion mailing list, and found out that GNU Radio Companion has a cache (see this post: https://lists.gnu.org/archive/html/discuss-gnuradio/2021-01/msg00181.html) under `$HOME/.cache/grc_gnuradio`; I am wondering if you could try to move that directory to a different name to see if the next time GRC runs it creates a new one with the right block file.

Also I am running GNU Radio 3.10 here too so the GNU Radio version shouldn't be a factor.

Franco

flocentblack, 2022-01-29:
That did it!  Works great, even when rapidly toggling bands. Thank you so much! I'll be watching out for the LNA state changes whenever you get the chance to work on that. Have a great weekend!

73,

Eric

closed 2022-01-29 by flocentblack