fventuri/gr-sdrplay3 #22: Problem building gr_sdrplay3

sksail92, 2022-02-07:
Hello Franco,
I am trying to build gr_sdrplay3 on a Raspberry pi. It seems to be tricky as I do have to build almost everything from srcatch as the 
packages I can download via APT are quite old. 
Nevertheless afte a while I managed to build gnuradio (and a lot of dependencies boost  gnuradio  gr-osmosdr  gr-sdrplay3  pybind11  pygccxml  SoapySDR  SoapySDRPlay  spdlog  volk) from the git repos. 

Only gr-sdrplay3 throws errors during Cmake and I have no idea why.  The build process seems to be quite complicated and 
I have no Idea where to start.
Maybe You are able to give some hints.
Kind regards
Harald

I do get the following error messages running "cmake .." from the build directory:

...
-- Configuring done
CMake Error at /usr/local/share/cmake/pybind11/pybind11Tools.cmake:165 (add_library):
  Target "sdrplay3_python" links to target "Python::Module" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/local/lib/cmake/gnuradio/GrPybind.cmake:260 (pybind11_add_module)
  python/bindings/CMakeLists.txt:32 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/local/share/cmake/pybind11/pybind11Tools.cmake:165 (add_library):
  Target "sdrplay3_python" links to target "Python::Module" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/local/lib/cmake/gnuradio/GrPybind.cmake:260 (pybind11_add_module)
  python/bindings/CMakeLists.txt:32 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:42 (add_library):
  Target "gnuradio-sdrplay3" links to target "Python::Module" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at lib/CMakeLists.txt:42 (add_library):
  Target "gnuradio-sdrplay3" links to target "Python::Module" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly._

fventuri, 2022-02-08:
@sksail92
I did build GNU Radio (version 3.10) and this `gr-sdrplay3` OOT module (from the `gnuradio-3.10` branch) on my Raspberry Pi 4 a few times (both 32 bit and 64 bit), and I've never seen error messages like yours; at first glance they make me think that something is wrong somewhere with the Pybind11 installation, but I can't really tell exactly what is the problem.

I think it might be helpful to show you my steps (more or less) so perhaps you can give it a try too:

- I installed Raspberry OS from the official Raspberry Pi downloads; set up IP address, ssh, etc
- I ran `apt update && apt upgrade` and rebooted the Raspberry Pi
- I downloaded and installed the SDRplay API for ARM processors (32 bit or 64 bit, depending on the architecture of the Raspberry OS you chose) from the SDRplay downloads: https://www.sdrplay.com/downloads/
- I cloned and built volk using the steps here for the 32 bit installation: https://github.com/gnuradio/volk#building-on-raspberry-pi-and-other-arm-boards-32-bit (you don't need those special instructions for 64 bit installation)
- To build GNU Radio 3.10 I followed the steps here: https://wiki.gnuradio.org/index.php?title=InstallingGRFromSource_on_Raspberry_Pi and here: https://wiki.gnuradio.org/index.php?title=InstallingGR#For_GNU_Radio_3.10,_3.9,_and_Main_Branch - notice the pybind11 is already available as a Raspberry OS package, and therefore there's no need to compile it from source (same thing for boost and spdlog). This step was the one that took the longest: about 2 hours (using `make -j3`) on my Raspberry Pi 4
- Finally I cloned and built this `gr-sdrplay3` OOT module (`gnuradio-3.10` branch) following the instructions in the README (https://github.com/fventuri/gr-sdrplay3#build-and-installation)
- For SoapySDR and the SoapySDRPlay module I followed the build instructions on their wiki (https://github.com/pothosware/SoapySDR/wiki/BuildGuide and https://github.com/pothosware/SoapySDRPlay3/wiki#building-soapy-sdr-play)

I hope this helps,
Franco

sksail92, 2022-02-14:
Hello Franco,
with your help I was finally able to build gr_sdrplay3.  The main problem was using the "master" branch. Than there were a lot of Pyton libs missing. I think that the "openplotter" software reconfigured the Raspberry in an odd way.

In the end I needed to add a lot of information in the cmake call to make it work.

cmake -DENABLE_PYTHON=1 -DCMAKE_BUILD_TYPE=Release\
-DPYTHON_INCLUDE_DIR=$(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") \
-DPYTHON_LIBRARY=$(python3 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") \
-DPYTHON_EXECUTABLE=/usr/bin/python3 \
../

Thanks again for Your help. 

My goal using gr-sdrplay3 with an sdrplay RspDx is to receive Navetx, Weatherfax and weather reports via RTTY on a boat.
If possibe all three at the same time as the stations are all located on the long- and medium wave band.

fventuri, 2022-02-15:
@sksail92 - I honestly am surprised that you had to add cmake options like `ENABLE_PYTHON`, which should already be set to `ON` in `GnuradioConfig.cmake` (which I think comes from `GrPython.cmake`: https://github.com/gnuradio/gnuradio/blob/main/cmake/Modules/GrPython.cmake) or `CMAKE_BUILD_TYPE=Release`, which should already be set here: https://github.com/fventuri/gr-sdrplay3/blob/gnuradio-3.10/CMakeLists.txt#L24

To decode NAVTEX in GNU Radio you might find useful the GNU Radio OOT module `gr-navtex` I wrote: https://github.com/fventuri/gr-navtex

Best of luck with your interesting project!
Franco