fventuri/gr-sdrplay3 #23: Gain controls for LNA state

fventuri, 2022-02-20:
@flocentblack
As we discussed a few weeks ago, I just created a new branch called `lnastate_gain_control` (https://github.com/fventuri/gr-sdrplay3/tree/lnastate_gain_control) that allows direct control of the LNA state (instead of having to go though the RF gain reduction).

In order to use it, I created a new gain name called `LNAstate` so that you can now assign directly the LNA state by using something like this in Python:
```
    set_gain(3, 'LNAstate')
```

Its equivalent in C++ would be something like this:
```
    set_gain(3, "LNAstate");
```

I also modified the GNU Radio Companion blocks so that if you use an RF gain that begins with the prefix 'LNA' (for instance 'LNA3'), it will invoke `set_gain(value, 'LNAstate')` - you can find an example of this way of passing the LNA state in the example `rspduo_dual_tuner_diversity_reception-LNA.grc` (https://github.com/fventuri/gr-sdrplay3/blob/lnastate_gain_control/examples/rspduo_dual_tuner_diversity_reception-LNA.grc).

Please give it a try when you have some time, and let me know if it works for you.

Franco

flocentblack, 2022-02-26:
Hi Franco,
I just tried the new branch. Works well when entering LNAx directly into the block. Still trying to figure out the correct syntax to enter LNArf_gain_slider_value into the RF gain field.

Thanks again!

fventuri, 2022-02-26:
@flocentblack - thanks for trying it out!

If you don't mind attaching the .grc file you are working on, I'll be happy to take a look too to help you out with the `LNArf_gain_slider_value`.

Franco

flocentblack, 2022-02-26:
Great! Here it is:
[nq8j_rx_2_6.txt](https://github.com/fventuri/gr-sdrplay3/files/8146713/nq8j_rx_2_6.txt)

fventuri, 2022-02-26:
@flocentblack - I just pushed a commit to the `lnastate_gain_control` branch to set the RF gain name to 'LNAstate' whenever the string 'LNA' (or 'lna') is in the RF gain field value; please download it and build with the latest changes.

I also added a couple of auxiliary variables (`lna_rx0` and `lna_rx1`; note that 'lna' is part of their name) in your GRC flowgraph, and I made these variables dependent on the LNA slider values; I then used these two auxiliary variables to set the RF gains in the RSPduo block.

Take a look at the attached .grc file and it should all make sense.

Franco

[nq8j_rx_2_6.zip](https://github.com/fventuri/gr-sdrplay3/files/8146902/nq8j_rx_2_6.zip)

flocentblack, 2022-03-01:
Wow. Thanks, I understand what you did to the block and changes to my flowgraph.

I tried the new branch and flowgraph you created. I'm getting these errors:
`  File "/home/supereb/Downloads/nq8j_rx_2.py", line 646, in <module>
    main()
  File "/home/supereb/Downloads/nq8j_rx_2.py", line 624, in main
    tb = top_block_cls()
  File "/home/supereb/Downloads/nq8j_rx_2.py", line 261, in __init__
    self.sdrplay3_rspduo_0.set_gain(-float(lna_rx0), -float(lna_rx1), 'RF')
ValueError: could not convert string to float: 'LNA3'`

When I attempted to clear the .cache, I noticed it was owned by root and had no contents.  I recently upgraded to MX-21, which may be why.

fventuri, 2022-03-02:
@flocentblack 
Eric, can you check what you have in lines 38-42 of the RSPduo GRC block (https://github.com/fventuri/gr-sdrplay3/blob/lnastate_gain_control/grc/sdrplay3_rspduo.block.yml#L38-L42)?

Those lines should look like this:
```
    % if 'LNA' in str(rf_gR0).upper() and 'LNA' in str(rf_gR1).upper():
    self.${id}.set_gain(int(${rf_gR0}[3:]), int(${rf_gR1}[3:]), 'LNAstate')
    % else:
    self.${id}.set_gain(-float(${rf_gR0}), -float(${rf_gR1}), 'RF')
    % endif
```

As you can see, if the RF gain fields contain 'LNA' or 'lna', that code should generate the Python line that invokes `set_gain()` with the gain name set to `LNAstate`.

If that block is correct the generated Python script should be very similar to mine (attached below) - notice for instance line 261:
```
    self.sdrplay3_rspduo_0.set_gain(int(lna_rx0[3:]), int(lna_rx1[3:]), 'LNAstate')
```

If yours has still the old calls to `set_gain(..., 'RF')`, it means there's something that is still using the old GRC block template.

Franco

[nq8j_rx_2.zip](https://github.com/fventuri/gr-sdrplay3/files/8165763/nq8j_rx_2.zip)

flocentblack, 2022-03-02:
In `/usr/local/share/gnuradio/grc/blocks/sdrplay3_rspduo.block.yml`, 38-42:
` % if str(rf_gR0)[1:4] == 'LNA' and str(rf_gR1)[1:4] == 'LNA':
    self.${id}.set_gain(int(${rf_gR0}[3:]), int(${rf_gR1}[3:]), 'LNAstate')
    % else:
    self.${id}.set_gain(-float(${rf_gR0}), -float(${rf_gR1}), 'RF')
    % endif`
Line 261 of generated python script:
` self.sdrplay3_rspduo_0.set_gain(-float(lna_rx0), -float(lna_rx1), 'RF')`

So it looks like the old version is cached somewhere.

fventuri, 2022-03-02:
No, this time is not a cache problem; if you look closely your line 38 is:
```
    % if str(rf_gR0)[1:4] == 'LNA' and str(rf_gR1)[1:4] == 'LNA':
```
while line 38 in the code in GitHub  (https://github.com/fventuri/gr-sdrplay3/blob/lnastate_gain_control/grc/sdrplay3_rspduo.block.yml#L38) is:
```
    % if 'LNA' in str(rf_gR0).upper() and 'LNA' in str(rf_gR1).upper():
```

Perhaps you forgot to run `git pull` or `sudo make install` from the latest version?

Franco

flocentblack, 2022-03-02:
Although I downloaded the latest version, I made the mistake of using the previous version with the same name located in my /home folder. Obviously, I'm 'amateur' at more than just radio...  Sorry that took up your time.

It works great. Thank you so much for implementing this, I appreciate it very much!

73,

Eric

fventuri, 2022-03-03:
Eric,
thanks for checking; I am glad to hear that everything works as expected.

I just merged these changes for the `LNA state` RF gain control into the `master` branch (and in the `gnuradio-3.10` branch).

73,
Franco

closed 2022-03-03 by fventuri