fventuri/gr-sdrplay3 #31: RSPDuo dual tuner independent rx - center_frequency in tuner 2 always identical to tuner 1

jcfitzpatrick12, 2024-02-23:
Hi Franco,

Firstly, thank you so much for your work on this repository, it has been hugely helpful for the project I am working on.

I have spent the evening trying to resolve an issue regarding the RSPDuo in dual tuner (independent rx) mode. I noticed that for a very simple flow graph:

[image]

which connects each output directly to a waterfall plot, that even when tuner 2 is set to a different frequency, it will be picking up an identical signal to that set for tuner 1.

With my antenna plugged in to tuner 1:
[image]

and tuner 2:
[image]

What's odd is that this is a very similar problem raised before: https://github.com/fventuri/gr-sdrplay3/issues/8 , which you have already resolved! I am using the latest version of the API 3.14.0, however, I was running into the same issue using an older version 3.07.1. I have also verified I am using the latest version of the code from this repository, with the fix you listed in the previous issue.

Any guidance on how I can solve this is much appreciated.

jcfitzpatrick12, 2024-02-23:
I have also noticed that as with https://github.com/fventuri/gr-sdrplay3/issues/8 , updating the frequency of tuner 2 in real time resolves the issue.

fventuri, 2024-02-23:
@jcfitzpatrick12 - thanks for the kind words about this project - I am really glad you find it useful!

I strongly suspect that the issue is with the API function `sdrplay_api_Init()`, which if I remember correctly resets the settings for tuner B to be the same same as for tuner A when the RSPduo runs in dual tuner mode.
I think I saw the same behavior when I was experimenting with the configuration of the RSPduo a couple of years ago (https://github.com/fventuri/dual-tuner-experiments).

To confirm this hypothesis, I would ask you to run this test:
- in the file 'rsp_impl.c' please uncomment line 522 (https://github.com/fventuri/gr-sdrplay3/blob/main/lib/rsp_impl.cc#L522) like this:
```
    print_device_config();
```
- also insert a line with the same call to `print_device_config();` a few lines below after the call to `sdrplay_api_Init()`, for instance right before line 545 (the line that begins with `run_status =`)

After these changes, please rebuild and reinstall the `gr-sdrplay3` module, and re-run your GNU Radio Companion flowgraph.

You should see on the terminal a detailed list with all the settings both before and after the call to `sdrplay_api_Init()`.
Please post it there so we can take a look and see if this is really what's happening in your case.

Franco

fventuri, 2024-02-24:
@jcfitzpatrick12 - thanks for running that test.

As you can see the results confirm what I was suspecting:
  - in the first 'Device config' output (before the function `sdrplay_api_Init()` is called), you can see that for instance for RX channel A `rfHz=9.58e+07` and for RX channel B `rfHz=5.4e+07`
  - in the second 'Device config' output (right after the function `sdrplay_api_Init()` has been called), you can see that RX channel A has `rfHz=9.58e+07` but RX channel B has `rfHz=9.58e+07` too (and many other settings have been also copied from channel A to channel B)

This morning I modified the `start()` function so that for the RSPduo in dual tuner mode it saves all the settings for channel B, runs `sdrplay_api_Init()`, and then reapplies back all the original settings for channel B (including `rfHz`), similar to what I had to do for  back then the program`dual_tuner_recorder` (https://github.com/fventuri/dual-tuner-experiments).

You'll find the updated code in the branch `rspduo-dual-tuner-mode-set-tuner-b`  (https://github.com/fventuri/gr-sdrplay3/tree/rspduo-dual-tuner-mode-set-tuner-b); please give it a try and let me know how it goes.

Also while running a quick test with your GNU Radio Companion flowgraph, I noticed that you are using a sample rate of 300ksps, which is not valid for the RSPduo in dual tuner mode (the only valid values are 2Msps, 1Msps, 500ksps, 250ksps, 125ksps, and 62.5ksps); I had to change it to 500ksps in order to run the flowgraph without errors.

Franco

jcfitzpatrick12, 2024-02-24:
Hi Franco,

Brilliant, that has fixed it got everything up and running as intended. Really appreciate your swift responses, this has been a big help :)

Jimmy

fventuri, 2024-02-24:
Jimmy,
glad to hear it worked for you - I just merged the updated code for the RSPduo in dual tuner mode to the `main` branch.

Franco

jcfitzpatrick12, 2024-02-25:
Hi Franco,

I just changed my container to run cloning off the ```main``` branch as opposed to ```3.9-fix_tuner2_frequency_for_independent_rx``` and noticed some odd issues in my gnuradio and on my application. 

If this is "my-end" issues, happy to resolve, but just thought to notify you as the issues were not present to the date of my last comment before the merge to main.

----
gnuradio issue
----

I am considering the following flowgraph, where tuner one is set to 95.8MHz (centered on local radio station) and tuner two is set to 95.9MHz (slightly displaced from local radio station).

[flow_graph.txt](https://github.com/fventuri/gr-sdrplay3/files/14395804/flow_graph.txt)

With the antenna plugged in to tuner 1:

[image]

With the antenna plugged in to tuner 2:

[image]

The frequency shift is as expected, but there is some odd filtering happening on tuner two.

-----
issue in application
-----
This seems related to the same problem as above. Source code can be found in ```grso``` repo on my page, but I am essentially just streaming the raw IQ from both tuners into two separate file meta sinks, then converting into radio spectrograms and plotting (can give more details if you need it). Data tagged with the flag "03" as in the figure is collected with the following parameters:

{
    "center_freq": 95.8e6,
    "samp_rate": 2e6,
    "IF_gain": -25,
    "RF_gain": 0,
    "average_over_int": 10
}

For the first figure I plug the tuner into tuner 1, and tell the Duo to collect with parameters for flag 03 as above. I tell tuner 2 (with nothing plugged in) to use some other set of configuration variables. The RSPDuo is running in independent rx mode. 

Output from tuner 1 with configuration for flag "03" as above:

[image]

An identical test is run except with the antenna plugged in to tuner 2. The output from tuner 2 with configuration for flag "03" as above:

[image]

At the date of my last comment, both tuners were giving identical results as expected.

As before, any guidance is much appreciated, apologies for the long issue.

Jimmy

fventuri, 2024-02-25:
@jcfitzpatrick12 - I agree that there's significant difference from what you can see coming in on tuner 1 and on tuner 2.
I haven't used the code in the `3-9-fix_tuner2_frequency_for_independent_rx` branch in a long time, and I suspect there are a lot of changes I added to the code since that branch.

A couple of things I would like you to try for me:
- please uncomment the call to `print_device_config()` in line 768 of `rspduo_impl.cc` (https://github.com/fventuri/gr-sdrplay3/blob/main/lib/rspduo_impl.cc#L768), rebuild, reinstall, and re-run your tests in the two cases. The detailed output in the terminal should give us a good idea of the settings of each tuner and perhaps we can spot an important difference there
- also I would like you to run a couple of tests (one for each tuner) with the same RSPduo in single tuner mode with the same sample rate, IF frequency (1620kHz), and IF bandwidth you used for dual tuner mode. Let me know if the results listening to the same range of frequencies are different between single tuner mode with tuner 1 and single tuner mode with tuner 2 (same antenna of course). These tests could help us understand if there is some other complexity introduced by dual tuner mode that is not present in single tuner mode.

Thanks for all your tests!
Franco

jcfitzpatrick12, 2024-02-25:
@fventuri 

Please let me know if any of the below needs clarification. I have ran the tests you suggested, and included all files in text attachments.

----- 

* _please uncomment the call to print_device_config() in line 768 of rspduo_impl.cc (https://github.com/fventuri/gr-sdrplay3/blob/main/lib/rspduo_impl.cc#L768), rebuild, reinstall, and re-run your tests in the two cases. The detailed output in the terminal should give us a good idea of the settings of each tuner and perhaps we can spot an important difference there_

I have pasted the terminal output of ```print_device_config()``` on running of the flowgraph in the previous comment into the following text file:

[gnuradio-test.txt](https://github.com/fventuri/gr-sdrplay3/files/14397577/gnuradio-test.txt)

Is it notable that the bwType variable decreases significantly [from 1536 to 200] moving from tuner A to tuner B? I notice that the bwType is identical for tuner A and B in the single tuner mode cases (see terminal outputs later in this comment).

I have also pasted the output of ```print_device_config()``` in the two tests I ran for my application. Firstly, the case where I tell the duo to collect with parameters for flag 03 from tuner 1, and tuner 2 (with nothing plugged in) to use some other set of configuration variables:

[grso-test-1.txt](https://github.com/fventuri/gr-sdrplay3/files/14397621/grso-test-1.txt)

and for the analogous test as in my previous comment where I swap the configurations round from tuner 1 to tuner 2, and vice versa:

[grso-test-2.txt](https://github.com/fventuri/gr-sdrplay3/files/14397627/grso-test-2.txt)

In both cases I checked visually the spectrograms and they were similar to my previous comment.

-----

_also I would like you to run a couple of tests (one for each tuner) with the same RSPduo in single tuner mode with the same sample rate, IF frequency (1620kHz), and IF bandwidth you used for dual tuner mode. Let me know if the results listening to the same range of frequencies are different between single tuner mode with tuner 1 and single tuner mode with tuner 2 (same antenna of course). These tests could help us understand if there is some other complexity introduced by dual tuner mode that is not present in single tuner mode_ 

For tuner 1 outputs are as expected:

flowgraph: 
[tuner_1_flowgraph.txt](https://github.com/fventuri/gr-sdrplay3/files/14397688/tuner_1_flowgraph.txt)
```print_device_config()``` output: 
[tuner_1_test_device_config.txt](https://github.com/fventuri/gr-sdrplay3/files/14397689/tuner_1_test_device_config.txt)
Waterfall plot:
[image]

For tuner 2, outputs are also as expected:
flowgraph:
[tuner_2_flowgraph.txt](https://github.com/fventuri/gr-sdrplay3/files/14397698/tuner_2_flowgraph.txt)
```print_device_config()``` output
[tuner_2_test_device_config.txt](https://github.com/fventuri/gr-sdrplay3/files/14397699/tuner_2_test_device_config.txt)

Waterfall plot:
[image]

fventuri, 2024-02-27:
@jcfitzpatrick12 - thanks for the detailed debug information.

I agree 100% with you that the bandwidth should be the same for both tuners. I also noticed a couple of other settings that should be the same but are different.

I have been busy with a few other things these days but tonight (or tomorrow night) after work I hope I have time to look at this problem a little better and try to understand what's going on.

Franco

fventuri, 2024-02-27:
@jcfitzpatrick12 - I think I figured out the problem.

I looked at what I was doing in the `3.9-fix_tuner2_frequency_for_independent_rx`  branch, and I realized that the right thing to do is to reapply to tuner B only those settings that are going to be different between two independent receivers in dual tuner mode, i.e. center frequencies and gains.

I modified my code to do just that, and it seems to be working correctly now.
You'll find these latest changes in the `main` branch; please give it a try and let me know if it works for you too.

Franco

jcfitzpatrick12, 2024-02-28:
Hi Franco, sorry for the delayed reply, was busy with work! Thanks for looking into this, I will check out everything now on my end and update you.

jcfitzpatrick12, 2024-02-28:
@fventuri delighted, everything's working as expected!

the spectrograms from tuner 1:
[image]
and tuner 2:
[image]
are both now showing identical outputs for the same configuration and connected to the same antenna.

Just a note, print_device_config() I think print_device_config() is still uncommented and is outputting to my terminal each time I am running (might have been intentional so I can check, but just letting you know just in case).

Thanks so much for your time and help :)

fventuri, 2024-02-29:
Jimmy, I 100% understand about work - the last few days have been very busy here too.

I am glad to see that dual tuner mode with independent receivers now works correctly, and thanks for reminding to comment out the function `print_device_config()` that I was using for debugging. It is fixed now in the `main` branch.

I'll wait a couple of days, and then I'll create a new tag in git with this latest commit, so people and distributions should start using the updated version.

Franco