fventuri/gr-sdrplay3 #32: Using two RSPduo at the same time, the raspberry pi 4b connection was unsuccessful

djl2020-ddd, 2024-02-26:
Hi, I'm using your program to run an experiment that requires at least two rspduo to run at the same time, but I got an error connecting to raspberry pi 4b using two rspduo, i have therefore come to you for a solution. 
I've seen questioners ask similar questions before, but since I'm not particularly familiar with all the RSPDUO, raspberry pie 4b, and gnuradio operations, I don't quite understand your solution. My version of gnuradio is 3.10.
thank you very much！
dingo-2020

djl2020-ddd, 2024-02-26:
我刚才又重新进行了测试，它在树莓派上运行成功了，但是由于某些原因我刚看到运行窗口的时候，程序崩溃了，然后我又在jetson orgin 上进行测试，测试结果报错：
Generating: '/home/jetson/桌面/dingo/top_block.py'

Executing: /usr/bin/python3 -u /home/jetson/桌面/dingo/top_block.py

rspduo :error: sdrplay_api_SelectDevice() Error: sdrplay_api_Fail
rspduo :error: sdrplay_api_GetDeviceParams() Error: sdrplay_api_NotInitialised

fventuri, 2024-02-27:
@djl2020-ddd - thanks for posting your message.

Unfortunately I only have one RSPduo here and I never tried with two.
If you post here your GNU Radio Companion flowgraph (the file with the '.grc' extension) I can take a look and see what could be the problem.

As per getting started with GNU Radio, there are several tutorials here: https://wiki.gnuradio.org/index.php?title=Tutorials
They also have a user discussion mailing list https://wiki.gnuradio.org/index.php/MailingLists and a chat on Matrix where you can ask questions directly to the developers and other users: https://wiki.gnuradio.org/index.php?title=Chat

Franco

djl2020-ddd, 2024-03-04:
Thank you very much for your response, and I seem to have an alternative solution, as I tried to do this experiment with multiple airspy R2 receivers, and thank you again for the advice that I will follow！