fventuri/gr-sdrplay3 #33: Building on headless pi5 / bookworm

BobSelf22, 2024-03-08:
I've installed SDRplay_RSP_API-Linux-3.14.0. That works.
I installed gnuradio from the bookworm repository.
gnuradio-config-info --version
3.10.5.1
so far, so good.

I built and installed gr-sdrplay3. I didn't notice any fatal errors.

When I run a no-gui python script it fails on this line:
from gnuradio import sdrplay3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/__init__.py", line 24, in <module>
    from .sdrplay3_python import *
ImportError: libgnuradio-sdrplay3.so.3.11.0git: cannot open shared object file: No such file or directory

Here's the output of the sudo make install:
				Install the project...
				-- Install configuration: "Release"
				-- Installing: /usr/local/lib/cmake/sdrplay3/sdrplay3Config.cmake
				-- Installing: /usr/local/include/gnuradio/sdrplay3/api.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rsp.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rsp1.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rsp1a.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rsp1b.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rsp2.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rspduo.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/rspdx.h
				-- Installing: /usr/local/include/gnuradio/sdrplay3/sdrplay3_types.h
				-- Installing: /usr/local/lib/aarch64-linux-gnu/libgnuradio-sdrplay3.so.v3.11-compat-xxx-xunknown
				-- Installing: /usr/local/lib/aarch64-linux-gnu/libgnuradio-sdrplay3.so.3.11.0git
				-- Set runtime path of "/usr/local/lib/aarch64-linux-gnu/libgnuradio-sdrplay3.so.v3.11-compat-xxx-xunknown" to ""
				-- Installing: /usr/local/lib/aarch64-linux-gnu/libgnuradio-sdrplay3.so
				-- Installing: /usr/local/lib/cmake/gnuradio-sdrplay3/gnuradio-sdrplay3Targets.cmake
				-- Installing: /usr/local/lib/cmake/gnuradio-sdrplay3/gnuradio-sdrplay3Targets-release.cmake
				-- Installing: /usr/local/lib/cmake/gnuradio-sdrplay3/gnuradio-sdrplay3Config.cmake
				-- Installing: /usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/__init__.py
				-- Installing: /usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/__init__.pyc
				-- Installing: /usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/__init__.pyo
				-- Installing: /usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/sdrplay3_python.cpython-311-aarch64-linux-gnu.so
				-- Set runtime path of "/usr/local/lib/python3.11/dist-packages/gnuradio/sdrplay3/sdrplay3_python.cpython-311-aarch64-linux-gnu.so" to ""
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rsp1.block.yml
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rsp1a.block.yml
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rsp1b.block.yml
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rsp2.block.yml
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rspduo.block.yml
				-- Installing: /usr/local/share/gnuradio/grc/blocks/sdrplay3_rspdx.block.yml
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/FM-Receiver-SDRplay.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rsp1a.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rsp1b.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rsp2.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rspduo_dual_tuner_diversity_reception-LNA.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rspduo_slave.grc
				-- Installing: /usr/local/share/gnuradio/examples/gr-sdrplay3/rspdx-LNA.grc

Maybe the repository version of gnuradio is not the 'main' branch?

Thank you

BobSelf22, 2024-03-08:
/usr/local/lib/aarch64-linux-gnu# ll
total 484K
lrwxrwxrwx 1 root root   33 Mar  8 13:59 libgnuradio-sdrplay3.so -> libgnuradio-sdrplay3.so.3.11.0git
lrwxrwxrwx 1 root root   49 Mar  8 13:59 libgnuradio-sdrplay3.so.3.11.0git -> libgnuradio-sdrplay3.so.v3.11-compat-xxx-xunknown
-rw-r--r-- 1 root root 482K Mar  8 13:58 libgnuradio-sdrplay3.so.v3.11-compat-xxx-xunknown

fventuri, 2024-03-08:
Thanks @BobSelf22 for reporting this issue.

Do you mind running this command to see if it fixes the error `cannot open shared object file: No such file or directory`?
```
sudo ldconfig
```

If the error is still there, could you run this command:
```
ldd /usr/local/lib/aarch64-linux-gnu/libgnuradio-sdrplay3.so.v3.11-compat-xxx-xunknown
```
and let me know if there's any `not found` line for the shared libraries it depends on?

Thanks,
Franco

closed 2024-08-06 by fventuri