fventuri/gr-sdrplay3 #35: ImportError: libgnuradio-sdrplay3.so.3.11.0git

acadiane, 2024-03-13:
Hello, 

I have Ubuntu 22.04
1.installAPI.sh
2.buildsoapy.sh
3.gr-sdrplay3 OOT

I can get my RSPduo with the API but I have an error with GNU Radio : 
File “/usr/local/lib/python3.10/dist-packages/gnuradio/sdrplay3/__init__.py”, line 24, in <module>
from .sdrplay3_python import *
ImportError: libgnuradio-sdrplay3.so.3.11.0git: cannot open shared object file: No such file or directory

I don't really understand the problem so I move the libgnuradio-sdrplay3.so.3.11.0git file to /usr/local/lib/python3.10/dist-packages/gnuradio/sdrplay3/, but the problem remains the same.

Do you have some advice ?

fventuri, 2024-03-14:
@acadiane - your error message is very similar to the one @BobSelf22 posted here: https://github.com/fventuri/gr-sdrplay3/issues/33

I suggested him to run the command:
```
sudo ldconfig
```
and it resolved his problem, so perhaps the same command will help with yours too.

Franco

acadiane, 2024-03-14:
Thanks a lot ! This command resolved the error :)