fventuri/gr-sdrplay3 #37: Cmake issue

ghassel, 2024-03-19:
I'm trying to install gr-sdrplay3 on Ubuntu 22.04, and encounter the following error when I run cmake:

cmake ..
-- Build type not specified: defaulting to release.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.65") found components: filesystem program_options regex thread 
-- Using GMP.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
-- Extracting version information from git describe...
-- Found pybind11: /usr/include (found version "2.9.1")
-- Using install prefix: /usr/local
-- Building for version: v3.11.0.4-0-gcaab7f71 / 3.11.0git
-- No C++ unit tests... skipping
-- User set python executable /usr/bin/python3.10
-- PYTHON and GRC components are disabled
-- Configuring donecmake ..
-- Build type not specified: defaulting to release.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.65") found components: filesystem program_options regex thread 
-- Using GMP.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
-- Extracting version information from git describe...
-- Found pybind11: /usr/include (found version "2.9.1")
-- Using install prefix: /usr/local
-- Building for version: v3.11.0.4-0-gcaab7f71 / 3.11.0git
-- No C++ unit tests... skipping
-- User set python executable /usr/bin/python3.10
-- PYTHON and GRC components are disabled
-- Configuring done
CMake Error in CMakeLists.txt:
  IMPORTED_LOCATION not set for imported target "gnuradio::gnuradio-runtime"
  configuration "Release".

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

CMake Error in CMakeLists.txt:
  IMPORTED_LOCATION not set for imported target "gnuradio::gnuradio-runtime"
  configuration "Release".

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

I have re-installed gnuradio and all of the dependencies, I think.   I appreciate any suggestions. 
Thanks!
-George

fventuri, 2024-03-19:
@ghassel - George, I have never seen that error before, however I just ran this command on my system:
```
find /usr -name gnuradio-runtimeTargets-release.cmake | xargs grep -H IMPORTED_LOCATION
```
and I do see that it finds the file `gnuradio-runtimeTargets-release.cmake`, and that that file contains the cmake variable `IMPORTED_LOCATION_RELEASE`:
```
/usr/local/lib64/cmake/gnuradio/gnuradio-runtimeTargets-release.cmake:  IMPORTED_LOCATION_RELEASE "${_IMPORT_PREFIX}/lib64/libgnuradio-runtime.so.v3.11.0.0git-672-gc3aa5bec"
```

Can you try running the same command on your computer to see if you have that file too and if it contains that variable?

Franco

ghassel, 2024-03-20:
Hi Franco,

I get: find /usr/ -name gnuradio-runtimeTargets-release.cmake | xargs grep
-H IMPORTED_LOCATION
/usr/lib/x86_64-linux-gnu/cmake/gnuradio/gnuradio-runtimeTargets-release.cmake:
 IMPORTED_LOCATION_RELEASE
"${_IMPORT_PREFIX}/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.1.1"

fventuri, 2024-03-20:
George,
please run cmake with the `--debug-find` option:
```
cmake --debug-find ..
```
The output will be very verbose but hopefully it will tell us what's going on.

Franco

ghassel, 2024-03-20:
Hi Franco,

Here it is:

cmake --debug-find ..
Running with debug output on for the `find` commands.
-- Build type not specified: defaulting to release.
CMake Debug Log at
/usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:141
(find_package):

/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headersConfig.cmake

/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headers-config.cmake

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:215
(boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:84 (find_package)

CMake Debug Log at
/usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:141
(find_package):
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:215
(boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:84 (find_package)

CMake Debug Log at
/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.74.0/boost_filesystem-config.cmake:100
(find_package):

/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headersConfig.cmake

/usr/lib/x86_64-linux-gnu/cmake/boost_headers-1.74.0/boost_headers-config.cmake

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:141
(find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:258
(boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:84 (find_package)

CMake Debug Log at
/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.74.0/boost_filesystem-config.cmake:100
(find_package):
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:141
(find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:258
(boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:84 (find_package)

CMake Debug Log at
/usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:141
(find_package):

/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.74.0/boost_filesystemConfig.cmake

/usr/lib/x86_64-linux-gnu/cmake/boost_filesystem-1.74.0/boost_filesystem-config.cmake

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake:258
(boost_find_component)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:84 (find_package)
[... 984 lines cut ...]

  The item was not found.

Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindDoxygen.cmake:627 (_Doxygen_find_dot)
  docs/CMakeLists.txt:12 (find_package)

CMake Debug Log at docs/CMakeLists.txt:12 (find_package):
  find_package considered the following paths for Doxygen.cmake

    /usr/local/lib/cmake/gnuradio/FindDoxygen.cmake
    /home/scmeteors/Research/gr-sdrplay3/cmake/Modules/FindDoxygen.cmake

  The file was found at

    /usr/share/cmake-3.22/Modules/FindDoxygen.cmake

-- PYTHON and GRC components are disabled
-- Configuring done
CMake Error in CMakeLists.txt:
  IMPORTED_LOCATION not set for imported target "gnuradio::gnuradio-runtime"
  configuration "Release".

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

fventuri, 2024-03-20:
George,
thanks for the very detailed log.

I took a quick look at it, and I see several references to the cmake directory `/usr/local/lib/cmake/gnuradio`. On the other hand, when you ran the `find` command earlier, you found that `gnuradio-runtimeTargets-release.cmake` was located under `/usr/lib/x86_64-linux-gnu/cmake/gnuradio`.

I suspect that you might have two different installations of GNU Radio on your computer: one from Ubuntu packages, which usually install shared libraries and cmake configuration files under `/usr/lib`, and a second one perhaps from building GNU Radio from source, where the cmake configuration files end up typically under `/usr/local/lib'.

I wonder if these two different installs are creating some confusion for cmake on your computer. I would suggest to decide which one you want to keep, and remove all the files for the other one.

Franco

ghassel, 2024-03-20:
Hi Franco,
Thanks for the suggestion and for reading all the way though that.   Is
there an installation method that works better, or should it not matter?

fventuri, 2024-03-20:
George,
both methods have pros and cons.
Since I see that you are using Ubuntu 22.04, which I think is LTS, I think you might prefer to stay with the GNU Radio version that comes with Ubuntu (I think it is 3.10.1.1) - it is not the newest 3.10 version, but it is not too obsolete either.

If instead you are like me who like to be on the bleeding edge, and don't mind fixing an issue building it every once in a while, then you should definitely build it from the main branch in git. This is also true if you need any feature that are in the latest version of GNU Radio but not in the version that comes with Ubuntu.

This GNU Radio OOT module 'gr-sdrplay3' should build and run without errors with either installation.

Franco

ghassel, 2024-03-21:
Hi Franco,
I uninstalled gnuradio, then re-installed it and gr-sdplay3, and everything
is working now.    The error does seem to stem from two GR installations.

Many thanks for the assistance