fventuri/gr-sdrplay3 #40: Adjusting LNAstate

BobSelf22, 2024-05-11:
I'm trying to use a QT GUI Range (type int) to adjust RF Attenuation in a flowgraph.
I have the id for this variable defined as 'lna' and the default value is 0.
When I start the flowgraph it prints this in the console:

Traceback (most recent call last):
  File "nbfm_gui.py", line 329, in <module>
    main()
  File "nbfm_gui.py", line 307, in main
    tb = top_block_cls()
  File "nbfm_gui.py", line 145, in __init__
    self.sdrplay3_rspdx_0.set_gain(int('lna'[3:]), 'LNAstate')
ValueError: invalid literal for int() with base 10: ''

fventuri, 2024-05-12:
@BobSelf22 - the gr-sdrplay3 source blocks accept the RF attenuation as either a numeric value (which means gain reduction in dB) or as the string 'LNA' followed by a number (for instance 'LNA3'), which allows the user to select a specific LNA value (https://github.com/fventuri/gr-sdrplay3/blob/main/grc/sdrplay3_rspdx.block.yml#L317-L320).

In your case I suspect that the name of your variable ('lna') causes a conflict in the GRC code that tries to figure out how the user is selecting the RF gain reduction: https://github.com/fventuri/gr-sdrplay3/blob/main/grc/sdrplay3_rspdx.block.yml#L23-L27

Franco

BobSelf, 2024-08-05:
Test for closing this issue. Looking for the button...

closed 2024-08-06 by fventuri