fventuri/gr-sdrplay3 #42: No module named 'gnuradio.sdrplay3.sdrplay3_python'

gozillah, 2024-06-02:
After installing gr-sdrplay3 on **Windows 10** and running a simple GNU Radio test I get the following error message:

> Traceback (most recent call last):
>   File "C:\Users\Ich\miniforge3\envs\gnuradio\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 24, in <module>
>     from .sdrplay3_python import *
> ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "C:\Users\Ich\gr-examples\FM\FM_Receiver.py", line 26, in <module>
>     from gnuradio import sdrplay3
>   File "C:\Users\Ich\miniforge3\envs\gnuradio\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 28, in <module>
>     from .sdrplay3_python import *
> ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'
> 
> 
I could not find a module named sdrplay3_python anywhere in the Internet. Please help.

fventuri, 2024-06-02:
I use Linux 99% of the times here, and to be honest I haven't built/run this module under Windows for quite a while, but, if I remember correctly, when you install 'gr-sdrplay3', it should create a dll called 'sdrplay3_python.dll' (or something like that) and install it somewhere in your 'miniforge3' environment.
I am sorry I forgot the exact details, but I think if you look at the output of the installation of 'gr-sdrplay3', it might have a line that says where it copied that dll (at least it does on Linux for shared libraries).

I hope this helps,
Franco

gozillah, 2024-06-03:
Thanks for your very swift reply. I found a file `sdrplay3_python.cp311-win_amd64.pyd` dated 03.03.2024 19:20, but no .dll file with name `sdrplay3_python` or similar. 

I then downloaded your gr-sdrplay3 sources and tried to build. After having installed boost, pybind11, pkgconfig and doxygen I got a clean cmake run, but no `sdrplay3_python.dll` or `*python.dll` or `sdrplay3*.dll` was created.

fventuri, 2024-06-03:
@gozillah thanks for the info

I honestly don't remember very well the name of that file; I thought it was a dll but now I am starting to have some doubt.

At this point the best thing probably is for me to boot my computer under Windows, install conda/radioconda there, rebuild this GNU Radio module, and see what happens (the old computer where I did this a while ago is gone).

I'll give it a try one of these nights after work, or during the weekend when I have more free time.

Franco

gozillah, 2024-06-03:
That's very kind, thank you. 

I am suggesting to use Radioconda 26.7.2023 with Ryan Volz's "base" environment. It is very easy to install and runs very well on my Windows 10. The later Radioconda releases (including the recent 29.5.2024 release) do seem to have a problem with some QTgui blocks, see my comment at [Radioconda issue 46](https://github.com/ryanvolz/radioconda/issues/46). 

May be you could also install Radioconda 29.5.2024 just for a test to confirm or contradict my findings.

gozillah, 2024-06-06:
Here [link](https://sdrplayusers.net/forums/topic/problem-sdrplay-with-gnu-radio-3-10-1-installed-via-ubuntu-22-04/) somebody seems to have had a similar problem under Ubuntu. However the solution of RabbitEars does not apply to Windows.

I tried your "Troubleshooting:" hint according to [link](https://groups.io/g/SDRPlayUsers/topic/howto_gnu_radio_on_windows/92079789) , but to no avail.

fventuri, 2024-06-07:
@gozillah - tonight I was able to build the conda package for Windows for this 'gr-sdrplay3' GNU Radio module, and I just posted a message about it here: https://github.com/fventuri/gr-sdrplay3/issues/41#issuecomment-2153777230

Please give it a try and let me know how it goes,
Franco