fventuri/gr-sdrplay3 #44: example rspdx.grc files Replace star characters in `rsp_selector` parameter with empty string in example

Abdull, 2024-06-14:
https://github.com/fventuri/gr-sdrplay3/blob/2fe5e830fe7a829529fbb08352a1e3ad824ee6c9/examples/rspdx.grc#L152

Currently, in example `rspdx.grc`, `rsp_selector` has the following value:

```
rsp_selector: '**********'
```

Because of this, running this example with e.g. `gnuradio-companion /usr/local/share/gr-sdrplay3/examples/rspdx.grc` will fail with 

```
Generating: '/tmp/rspdx.py'

Executing: /usr/bin/python3 -u /tmp/rspdx.py

rspdx :error: SDRplay device not found: **********
Traceback (most recent call last):
  File "/tmp/rspdx.py", line 208, in <module>
    main()
  File "/tmp/rspdx.py", line 186, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "/tmp/rspdx.py", line 85, in __init__
    self.sdrplay3_rspdx_0 = sdrplay3.rspdx(
                            ^^^^^^^^^^^^^^^
RuntimeError: sdrplay device not found or invalid mode/antenna selection

>>> Done (return code 1)
```

fventuri, 2024-06-15:
@Abdull - thanks for reporting this problem.

It should be fixed now.

Franco

Abdull, 2024-06-15:
Awesome! I rebuild gr-sdrplay with your fixes introduced in https://github.com/fventuri/gr-sdrplay3/commit/0b11cf3a389df48c66eb6dd0d4f8449e6cd81ac4 . Running `gnuradio-companion /usr/local/share/gr-sdrplay3/examples/rspdx.grc` now works directly out of the box without modifications. Great experience, thank you very much for *gr-sdrplay3*!

closed 2024-06-15 by Abdull