fventuri/gr-sdrplay3 #46: 'gnuradio.sdrplay3' has no attribute 'rspdxr2'

JoeTester1965, 2024-06-22:
Fresh install Ubuntu 22.04, gnuradio and SDRPlay V1.3 API.

Get this error for example RSPdx-R2 block (SDRConnect app works ok and API is up and running)

Traceback (most recent call last):
  File "/home/jetsondev/Documents/gr-sdrplay3/examples/rspdx_r2.py", line 207, in <module>
    main()
  File "/home/jetsondev/Documents/gr-sdrplay3/examples/rspdx_r2.py", line 185, in main
    tb = top_block_cls()
  File "/home/jetsondev/Documents/gr-sdrplay3/examples/rspdx_r2.py", line 84, in __init__
    self.sdrplay3_rspdxr2_0 = sdrplay3.rspdxr2(
AttributeError: module 'gnuradio.sdrplay3' has no attribute 'rspdxr2'. Did you mean: 'rspdx'

Thanks for any help you can give.

fventuri, 2024-06-22:
@JoeTester1965 - good catch - when I added the code for the new RSPdx-R2, I forgot to add the call to the function with the Python bindings for it, so the bindings were there but they were never being called.

I just pushed a commit with the additional line to the `main` branch.

Please run `git pull` again, and this time it should build and run without any errors.

Franco

JoeTester1965, 2024-06-22:
Thanks, got there but had to manually edit  ./python/sdrplay3/bindings/python_bindings.cc and add  

bind_rspdxr2(m)

to PYBIND11_MODULE(sdrplay3_python, m)

fventuri, 2024-06-23:
@JoeTester1965 - that line you manually added is exactly what I had in the commit I pushed to the `main` branch: https://github.com/fventuri/gr-sdrplay3/commit/5165bbdcb99544414084f1cf6b78113fda21e1a0
I am not sure if you had to add it manually after running `git pull` again, or if you added it instead of running `git pull` the second time.

Franco

JoeTester1965, 2024-06-23:
Hi - This change was needed *after* a git pull, need to change python binding calls as well as prototypes, so two changes are needed in the .cc file. Joe.

JoeTester1965, 2024-06-23:
Thanks for this module, appreciated. Joe.

closed 2024-06-23 by JoeTester1965

fventuri, 2024-06-23:
Thanks for kind words and good catch again.
I added that missing line to the `main` branch, so now it should work without needing any manual changes.

Franco