fventuri/gr-sdrplay3 #47: Do these blocks support remote operation?

JoeTester1965, 2024-06-24:
Hi, not urgent but was wondering if these blocks support connection to a remote sdrplay device running as a server.

Have tried SoapyRemote but am having issues getting that working as documented here:

https://github.com/pothosware/SoapyRemote/issues/106

fventuri, 2024-06-24:
Joe,
GNU Radio blocks (including the ones in the 'gr-sdrplay3' OOT module) are usually designed to run inside a GNU Radio flowgraph running on some computer.
What people usually do with GNU Radio when they need a functionality similar to SoapyRemote, i.e. the ability to run part of the sourcing and processing on a remote computer and another part of the processing on a local computer, is to use ZeroMQ (ZMQ) blocks: https://wiki.gnuradio.org/index.php/Understanding_ZMQ_Blocks

For instance in your case you could write a remote flowgraph (likely in 'No GUI' mode) that has a SDRplay RSPdx-R2 source, perhaps some initial preprocessing (to say limit the data being sent over the network), and a ZMQ sink block.
Then on a local computer you would have another flowgraph, with a ZMQ source block and the rest of your processing. This latter one would use the QT GUI, so you could have GUI controls, QT frequency sinks, etc.

With ZMQ blocks you have the flexibility to create different flowgraph structures (for instance you could have two ZMQ sink blocks and therefore be able to send your data to two different computers), while SoapyRemote is more targeted to a specific use case.

Franco

JoeTester1965, 2024-06-24:
Thanks for that, looks a sensible route!