fventuri/gr-sdrplay3 #49: ImportError: generic_type: type "rsp" referenced unknown base type "gr::sync_block"

atmci, 2024-07-07:
### Background
System: Windows 11 64bit. 
SDR device: SDRplay RSPdx 

Radioconda installer (radioconda-2024.05.29-Windows-x86_64.exe) downloaded from: 
https://github.com/ryanvolz/radioconda 
(GNU Radio Companion working ok.) 

SDRplay API 3.15 (SDRplay_RSP_API-Windows-3.15.exe) downloaded from: 
https://www.sdrplay.com/api/
(SDRuno and SDRconnect working ok.) 

GNU Radio OOT module for SDRplay RSP devices 
(gnuradio-sdrplay3-3.11.0.2.post2+ga4071c6-py311hf0272db_0.tar.bz2) 
downloaded and installed from: 
https://github.com/fventuri/gr-sdrplay3/releases 

Example flow graph file (rspdx.grc) downloaded from: 
https://github.com/fventuri/gr-sdrplay3/tree/main/examples
(Flow graph displayed ok in GNU Radio Companion.) 

Python file (rspdx.py) created ok.

### Error Messages
However, when executing the flow graph the following errors are reported: 
~~~
ImportError: generic_type: type "rsp" referenced unknown base type "gr::sync_block"
ImportError: generic_type: type "stream_args" is already registered!
~~~

The full messages are:
~~~
Executing: C:\Users\Alast\radioconda\python.exe -u C:\Users\Alast\Documents\GNU Radio\rspdx.py

Traceback (most recent call last):
  File "C:\Users\Alast\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 24, in <module>
    from .sdrplay3_python import *
ImportError: generic_type: type "rsp" referenced unknown base type "gr::sync_block"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Alast\Documents\GNU Radio\rspdx.py", line 25, in <module>
    from gnuradio import sdrplay3
  File "C:\Users\Alast\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 28, in <module>
    from .sdrplay3_python import *
ImportError: generic_type: type "stream_args" is already registered!

>>> Done (return code 1)
~~~

In addition, it produces the same error from a python prompt with: 
~~~
from gnuradio import sdrplay3
~~~

*I was really interested to hear about GNU Radio recently, and your module for SDRplay devices. 
It will be a great way to demonstrate to school pupils how radio signals are processed. 
So, a big thank you for making this available.*

Alastair

fventuri, 2024-07-07:
@atmci - thanks for the kind words. I really hope this project can help your students!

To be honest I use Linux most of the time (that's what I run on this computer to develop this GNU Radio OOT module) and I haven't rebuild it for radioconda in some time, so it is possible that one of the recent code changes could have caused that error.

One night this week after work I'll find some time to reboot my computer under Windows, rebuild the radioconda package for 'gnuradio-sdrplay3', install it, and try to run the simple statement `from gnuradio import sdrplay3` to see if I have the same  error message as you.
If I am able to reproduce it here, then I should be able to fix it quickly. If it works for me instead, then we can compare my installation of radioconda and the `gr-sdrplay3` module with yours, and hopefully we'll figure out the reason for the different behavior.

Franco

fventuri, 2024-07-08:
@atmci - I just rebuilt the Conda package `gnuradio-sdrplay3` with the latest code from the `main` branch (tag v3.11.0.6).

You can find the prebuilt package for Windows here: https://github.com/fventuri/gr-sdrplay3/releases/tag/v3.11.0.6

I also ran the simple test running the `from gnuradio import sdrplay3` statement in the Conda Python REPL (in Windows 11), and it ran without any error:
 ```
(base) C:\Users\franco\workdir\gnuradio>python
Python 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:27:10) [MSC v.1938 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from gnuradio import sdrplay3
>>>
```

Franco

atmci, 2024-07-08:
Hi Franco, 

v3.11.0.6 works fine. 
Many thanks for your very fast response! 

Alastair

closed 2024-07-08 by atmci

John-Jobin-McAuliffe, 2024-07-31:
I'm getting the same error using the same windows version, have tried the different versions of the package and the error is persisting with the Conda install. Not sure what is wrong.

fventuri, 2024-07-31:
@John-Jobin-McAuliffe - thanks for reporting this problem.

Do you have perhaps multiple version of conda (or say PothosSDR) installed on your computer?

The base type `gr::sync_block` should be provided by your GNU Radio installation (via radioconda), so I was wondering if there's any conflict somewhere, or perhaps one of the packages is at the wrong version (see here for instance: https://lists.gnu.org/archive/html/discuss-gnuradio/2021-05/msg00057.html).

To try to solve this issue, I would make a full backup of your current radioconda installation, then uninstall radioconda, make sure you do not have any leftover radioconda*, miniforge3* or PothosSDR folders, then reinstall radioconda 2024.05.29, and finally this OOT module `gr-sdrplay3`.

Franco

John-Jobin-McAuliffe, 2024-08-01:
@fventuri Hi Franco,

Ended up doing a systematic uninstall and reinstall of everything, gnu, conda-forge, etc and then reinstalled the packages and your OOT module as you said and it worked! Windows is bizarre but I'm glad that it is properly installed now. Thank you so much for the help!

John