fventuri/gr-sdrplay3 #50: Running on a pi5

BobSelf, 2024-08-05:
I recently switched operating systems on a laptop from Ubuntu to the current stable Debian. I cloned this repository yesterday. I built it for Debian on the Intel laptop. It works fine.  Since bookworm requires that I use a virtual environment if I want to install python modules, I had to set PYTHONPATH for the venv:
export PYTHONPATH=/usr/lib/python3/dist-packages
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.11/dist-packages

I also built this on the pi5 the same way.

When I run my python program on the pi5 it works up to the point when a set() function is called. I've tried:
set_center_freq(freq)
set_gain(-20.0, 'IF')

either of these calls give this exception or similar:

exception: set_center_freq(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.sdrplay3.sdrplay3_python.rsp, freq: float, synchronous: bool) -> float

Invoked with: <gnuradio.sdrplay3.sdrplay3_python.rsp1b object at 0x7ffefd14e7b0>, 1380000.0

I call 'getters' before this and they all work.

thank you,
Bob

fventuri, 2024-08-05:
@BobSelf - good catch.

A month or so ago I added the new parameter `synchronous` to some of the those set() methods (for sample rate, center frequency, and gains; see discussion here: https://github.com/fventuri/gr-sdrplay3/issues/45#issuecomment-2198376407).

I set the default for 'synchronous' to false (so it matches the current behavior) in the C++ code, but I completely forgot to do the same in the Python bindings.

I just pushed this commit https://github.com/fventuri/gr-sdrplay3/commit/d1351b125be1a2ed57a1569edf83c6b8eecb2481 to fix the problem with Python; I also ran a quick test here and it seems to work correctly.

Please give it a try and let me know if it works for you too.

Franco

BobSelf, 2024-08-05:
It seems to work fine now.

Thank you... I look forward to experimenting with gnuradio 'messages' (I think that's what I saw) on the 'synchrounous' discussion.

closed 2024-08-05 by BobSelf