fventuri/gr-sdrplay3 #52: SDRPlay devices troublesome gain

JohnNatHaysys, 2024-10-09:
I appreciate that GNSS-SDR  forum may be more appropriate to post this problem, but users here  and fventuri seem more likely to solve the problem.

GNSS-SDR has several RF front ends but the RSPdx is not one of them. I have loaded API 3.15, built the main branch of SoapySDR3 and the (current?) gr-osmosdr. The latter claims support for soapy. It works - almost.  If left long enough, all satellites eventually get detected but not enough at the same time. I suspect  there is a problem with setting the gain stages in the RSPdx. As I understand it, this device uses attenuators. My supposition is that the current configuration is overloading some stages, resulting in poor performance on low level signals. Simply attenuating the input does not help.

The osmosdr has rf and if gain parameters, but these have no effect. Is anyone aware of how to set the if and rf gains from osmosdr when it is interfacing with soapy.?

fventuri, 2024-10-09:
To be honest I am not familiar withe GNSS-SDR project, but I took a quick look this morning and I saw that it can also take a file with raw samples as the signal source.
To troubleshoot the problems with GNSS-SDR and the RSPdx, I think it might be useful initially to save the raw samples to file, and play that file with GNSS-SDR to see if it is able to decode enough satellites.

There are a few options to save the raw samples from your RSPdx to file.
For instance I wrote a simple utility program called `single_tuner_recorder` (https://github.com/fventuri/single-tuner-experiments) that allows you to do just that, and has controls for both RF and IF gains (actually RF and IF gain reductions, as per SDRplay gain model). Since it interfaces directly with the plain SDRplay API, it might be a good way to see if the problem comes from gr-osmosdr or SoapySDRPlay3.
Another option would be to build a simple GNU Radio flowgraph using this `gr-sdrplay3` source module and a file sink that writes to a file; in this case too you should be able to control both the RF and IF gains reliably.

Once we have a good idea of what's going on, we can perhaps use a pipeline to connect the two parts together instead of a file on disk.

Franco

JohnNatHaysys, 2024-10-09:
Hi,
Since posting the original I found that I needed to increase the signal to the RSPdx and then it worked correctly. However, by switching the input signal levels I could see that there is an odd AGC effect. Trying to turn off the AGC in the gr-osmosdr gave an SDR-API error about IFGR since AGC was on. Too many layers of drivers.
This seems an inevitable consequence of a closed source library. in any case there was no improvement in the performance over the RTL-SDR V3 for this application. Thanks for your comments though.

closed 2024-10-09 by JohnNatHaysys