fventuri/gr-sdrplay3 #53: Install module

phillinux94, 2024-11-12:
when sudo make install : 

CMake Error at python/sdrplay3/cmake_install.cmake:50 (file):
  file INSTALL cannot find
  "/home/philippe/Téléchargements/gr-sdrplay3/build/python/sdrplay3/__init__.pyc":
  No such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:55 (include)

fventuri, 2024-11-13:
@phillinux94 - I just ran a new build (with the usual cmake, make) followed by `sudo make install` here on my PC (running Linux Fedora), and it worked without errors.

I also took a quick look at the cmake generated file `build/python/sdrplay3/cmake_install.cmake`, and I don't see any reference to `init.pyc`.

If you don't mind, can you attach here your `build/python/sdrplay3/cmake_install.cmake` so I can compare it with mine to try to figure out the problem?

Franco

fventuri, 2024-11-15:
Thanks for your `cmake_install.cmake` file.
I just compared it with mine, and I see that in lines 49-54 you have:
```
if(CMAKE_INSTALL_COMPONENT STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib/python3.12/dist-packages/gnuradio/sdrplay3" TYPE FILE FILES
    "/home/philippe/Téléchargements/gr-sdrplay3/build/python/sdrplay3/__init__.pyc"
    "/home/philippe/Téléchargements/gr-sdrplay3/build/python/sdrplay3/__init__.pyo"
    )
endif()
```

while for the same case I just have:
```
if(CMAKE_INSTALL_COMPONENT STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib64/python3.13/site-packages/gnuradio/sdrplay3" TYPE FILE FILES "/home/franco/SDR/gr-sdrplay3/python/sdrplay3/__init__.py")
endif()
```

In your case you have references to the files `__init__.pyc` and `__init_.pyo`, while in my case I just have `__init__.py`

Perhaps there are some left overs from a previous build?

I would suggest to delete (or move somewhere else) the `build` directory, and rerun a fresh build of this GNU Radio OOT module as described here: https://github.com/fventuri/gr-sdrplay3?tab=readme-ov-file#build-and-installation 

Franco