fventuri/gr-sdrplay3 #57: Failed at last step of installation in Windows 11

JLRMSousa, 2025-04-15:
Installing gnuradio companion for the first time, following the procedure at [https://github.com/fventuri/gr-sdrplay3/blob/main/WINDOWS.md]

I am pasting here my whole sequence of commands in the windows powershell, with a lot of the positive response trimmed out with ... see the two failures highlighted at the end.

Not present in this sequence was the installation of the windows API for SDRplay in step 2 of the instructions above

Also, in step 1, there is no new "Conda Prompt" in the start menu. I have just used the windows command prompt power shell.

########################################################################
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.
Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows
PS C:\Users\me> cd C:\Users\me\miniforge3
PS C:\Users\me\miniforge3> cd condabin
PS C:\Users\me\miniforge3\condabin> .\conda.bat
usage: conda-script.py [-h] [-v] [--no-plugins] [-V] COMMAND ...
...
PS C:\Users\me\miniforge3\condabin> .\conda create -n gnuradio
Channels:
 - conda-forge
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done
...
## Package Plan ##
  environment location: C:\Users\me\miniforge3\envs\gnuradio
Proceed ([y]/n)? y
...
PS C:\Users\me\miniforge3\condabin> .\conda activate gnuradio
PS C:\Users\me\miniforge3\condabin> .\conda config --env --add channels conda-forge
Warning: 'conda-forge' already in 'channels' list, moving to the top
PS C:\Users\me\miniforge3\condabin> .\conda config --env --set channel_priority strict
PS C:\Users\me\miniforge3\condabin> .\conda install gnuradio
Channels:
 - conda-forge
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done
==> WARNING: A newer version of conda exists. <==
    current version: 24.11.3
    latest version: 25.3.1
Please update conda by running
    $ conda update -n base -c conda-forge conda
## Package Plan ##
  environment location: C:\Users\me\miniforge3
  added / updated specs:
    - gnuradio
The following packages will be downloaded:
...
The following packages will be UPDATED:
  openssl                                  3.4.1-ha4e3fda_0 --> 3.5.0-ha4e3fda_0
Proceed ([y]/n)? y
Downloading and Extracting Packages:
Preparing transaction: done
Verifying transaction: done
Executing transaction: -
/
/
done
PS C:\Users\me\miniforge3\condabin> .\conda activate gnuradio
PS C:\Users\me\miniforge3\condabin> gnuradio-companion            ##########FAILED HERE#######
gnuradio-companion : The term 'gnuradio-companion' is not recognized as the name of a cmdlet, function, script file,
or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and
try again.
At line:1 char:1
+ gnuradio-companion
+ ~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (gnuradio-companion:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
PS C:\Users\me\miniforge3\condabin> .\mamba update --all -y
Looking for: ['adwaita-icon-theme', 'archspec', 'atk-1.0', 'attrs', 'boltons', 'brotli', 'brotli-bin', 'brotli-python', 'bzip2', 'ca-
...
Downloading and Extracting Packages:
Preparing transaction: done
Verifying transaction: done
Executing transaction: |
/
done
PS C:\Users\me\miniforge3\condabin> .\mamba activate base
PS C:\Users\me\miniforge3\condabin> cd 'C:\_Photo Gallery_\EE\_SDR'
PS C:\_Photo Gallery_\EE\_SDR> cd 'C:\Users\me\miniforge3\condabin'
PS C:\Users\me\miniforge3\condabin> .\mamba install --offline "gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2"
critical libmamba Specified pkgs_dir does not exist             ##########FAILED HERE###############
########################################################################

After this I added "C:\Users\me\miniforge3\condabin>" to the windows 11 path and rebooted. Tried the last failed steps to no effect.

Very grateful in advance for any help.
Regards,
-Joe

fventuri, 2025-04-15:
@JLRMSousa - Joe, thanks for reporting this problem.

I normally use Linux here, but after reading your message I noticed that Ryan Volz published a new version of radioconda last month: [https://github.com/ryanvolz/radioconda/releases/tag/2025.03.14](https://github.com/ryanvolz/radioconda/releases/tag/2025.03.14).
I suspect there might be some changes that I have to make to my conda recipe to make it work with the latest radioconda.

I'll take a look at what needs to be changed in the new few days, and I'll let you know.

Also the default installation of radioconda here is under the folder 'radioconda', not 'miniforge3' (i.e. 'C:\Users\me\radioconda' in your case). Did you use a different way of installing it by any chance?

Franco

JLRMSousa, 2025-04-16:
Thank you so much, Franco for your prompt response!
I tried to follow the instructions at [](https://github.com/fventuri/gr-sdrplay3/blob/main/WINDOWS.md) as much as possible.
It sounds like I may need to uninstall what I have at 'C:\Users\me\minforge3' in order to end up with with the default installation of radioconda at 'C:\Users\me\radioconda'. Correct?

**For what it's worth I get these environments under conda:**
PS C:\Users\me\miniforge3\condabin> conda info --envs
...
base                   C:\Users\iinvn\miniforge3
gnuradio               C:\Users\iinvn\miniforge3\envs\gnuradio

**This is the info I get from conda:**
PS C:\Users\me\miniforge3\condabin> conda info
     active environment : None
       user config file : C:\Users\me\.condarc
 populated config files : C:\Users\me\miniforge3\.condarc
                          C:\Users\me\.condarc
          conda version : 25.3.1
    conda-build version : not installed
         python version : 3.12.10.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=25.3.1=0
                          __win=10.0.26100=0
       base environment : C:\Users\me\miniforge3  (writable)
      conda av data dir : C:\Users\me\miniforge3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\me\miniforge3\pkgs
                          C:\Users\me\.conda\pkgs
                          C:\Users\me\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\me\miniforge3\envs
                          C:\Users\me\.conda\envs
                          C:\Users\me\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/25.3.1 requests/2.32.3 CPython/3.12.10 Windows/11 Windows/10.0.26100 solver/libmamba conda-libmamba-solver/25.3.0 libmambapy/2.0.8
          administrator : False
             netrc file : None
           offline mode : False

Best regards,
-Joe

fventuri, 2025-04-16:
Joe,
perhaps you have a mix of installations of regular conda (under 'miniforge3') and radioconda (under 'radioconda').

The two installations might be able to coexist side by side since they use different top folders, but under your home directory the'll both create a folder called '.conda' where there's a file called 'environments.txt', so you may have to fix that to make both work.

Or perhaps it is safer and easier to choose one or the other; I chose radioconda because I use it only for testing my GNU Radio OOT module, but in your case you may have installed the standard conda for some other project or work, so it really depends.

Last night I found out that my Windows 11 installation (that I use probably once every few months) was too old (22H2) and no longer supported by Microsoft. I started the process of upgrading to the latest version, only to discover that my PC doesn't have TPM enabled, so now I am in the process of updating the BIOS to see it if helps.

I'll let you know how it goes.

Franco

fventuri, 2025-04-17:
Joe,
I am still fighting with TPM on my PC. Tonight I tried changing several settings in the BIOS, but I am still getting this error 'No security device found', and therefore I am still unable to upgrade Windows 11.

Anyway I just googled that error message and found this issue where they are having the same error message: https://github.com/mamba-org/mamba/issues/1368 - I thought I would mention it to you because perhaps you can find some useful information in that thread.

Franco

JLRMSousa, 2025-04-17:
I uninstalled miniforge3 in the original widows 11 PC and installed radioconda as I did in windows 10. I probably got confused with the windows 11 installation at the top of this thread.
 
This ran without incident and this time it generated the radioconda prompt icon. I opened the radioconda prompt and the gnuradio-sdrplay3 package installed without complaint. This is the command I used in the directory where I had downloaded the ...tar.bz2 package:
mamba install --offline "gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2"
A GNU radio companion is now present in the windows 11 start menu.

I opened it and loaded the FM Radio example for SDRplay with the RSPduo
C:\Users\me\radioconda\Library\share\gr-sdrplay3\examples\FM-Receiver-SDRplay.grc
I tried to run this and got the very common error: **The executable 'xterm' is missing**
I clicked the Options block within the FM-Receiver-SDRplay.grc flow graph and confirmed that my "Generate Option = QT GIU". The xterm is only relevant for the "NO GUI" options.

**However, running the flow graph again, I still don't get FM operation.**
**The following is what was reported in the radio companion console:**

Generating: "C:\Users\me\radioconda\Library\share\gr-sdrplay3\examples\top_block.py"

Executing: C:\Users\me\radioconda\python.exe -u C:\Users\me\radioconda\Library\share\gr-sdrplay3\examples\top_block.py

Traceback (most recent call last):
  File "C:\Users\me\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 26, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\me\radioconda\Library\share\gr-sdrplay3\examples\top_block.py", line 26, in <module>
    from gnuradio import sdrplay3
  File "C:\Users\me\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 30, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'

>>> Done (return code 1)
Regards,
-Joe

fventuri, 2025-04-18:
Joe, congratulations!
I think you are making very good progress there!

The message about 'xterm' is harmless - see here:  https://wiki.gnuradio.org/index.php?title=Talk:Your_First_Flowgraph

The other 'module not found' error is more serious. I just looked in my old installation of radioconda, and here I see the file `%CONDA_PREFIX%\Library\bin\gnuradio-sdrplay3.dll`, where `CONDA_PREFIX` is the name of the folder where you installed radioconda, i.e. `C:\Users\me\radioconda` in your case.

Do you have that dll too? If you do, does it have dependencies on other dll's it is not able to find?

Franco

JLRMSousa, 2025-04-18:
Thanks, Franco!
Yes, I have that dll in bin directory.
-Joe

JLRMSousa, 2025-04-18:
Franco, I attached the full file/directory tree under radioconda.
-Joe
[radioconda_tree.csv](https://github.com/user-attachments/files/19812895/radioconda_tree.csv)

fventuri, 2025-04-19:
Thanks Joe.
Tonight I found some useful information on how to troubleshoot that 'ModuleNotFoundError' error message with radioconda: https://wiki.gnuradio.org/index.php/CondaInstall#When_I_try_to_run_GNU_Radio,_why_do_I_get_a_ModuleNotFoundError_(no_module_named_...)?

One possible reason they mention there is a conflicting GNU Radio or Python installation.
They also suggest checking your Python path.

Franco

JLRMSousa, 2025-04-19:
Thanks for the info, Franco.
This is what I get from the python path:

(base) C:\Users\me>python -c "import sys; print('\n'.join(sys.path))"
C:\Users\me\radioconda\python312.zip
C:\Users\me\radioconda\DLLs
C:\Users\me\radioconda\Lib
C:\Users\me\radioconda
C:\Users\me\radioconda\Lib\site-packages
C:\Users\me\radioconda\Lib\site-packages\win32
C:\Users\me\radioconda\Lib\site-packages\win32\lib
C:\Users\me\radioconda\Lib\site-packages\Pythonwin

The first entry does not exist.
I searched for sys.path in the whole C: drive but only found things like sys_path.py
Where can I find sys.path to eliminate the first entry? Or should I get the missing python312.zip?
Regards,
-Joe

fventuri, 2025-04-21:
Joe,
I think Python syspath is just a list of places where your installation of radioconda Python looks for modules (conceptually similar to the environment variable PATH used by Windows to look for commands).
The fact that python312.zip is not there is normal (I don't think I have it here either); that just means that Python will not look for modules in that zip file.

By the way, did you already read and follow the troubleshooting instructions from the `WINDOWS.md` file in the `gr-sdrplay3` repository: https://github.com/fventuri/gr-sdrplay3/blob/main/WINDOWS.md#troubleshooting ?

Franco

n1ai, 2025-04-23:
I installed `gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2` and `SDRplay_RSP_API-Windows-3.15.exe` as Joe did.

When I run the command to print the sys.path from within the radioconda prompt I get:

```
(base) C:\Users\shaq\radioconda> python -c "import sys; print('\n'.join(sys.path))"

C:\Users\shaq\radioconda\python312.zip
C:\Users\shaq\radioconda\DLLs
C:\Users\shaq\radioconda\Lib
C:\Users\shaq\radioconda
C:\Users\shaq\radioconda\Lib\site-packages
C:\Users\shaq\radioconda\Lib\site-packages\win32
C:\Users\shaq\radioconda\Lib\site-packages\win32\lib
C:\Users\shaq\radioconda\Lib\site-packages\Pythonwin
```

We can see the gnuradio.sdrplay3 files under Lib\site-packages  are in the right place:

```
(base) C:\Users\shaq\radioconda>dir Lib\site-packages\gnuradio\sdrplay3
 Volume in drive C is Windows
 Volume Serial Number is 4849-C9E3

 Directory of C:\Users\shaq\radioconda\Lib\site-packages\gnuradio\sdrplay3

04/23/2025  07:05 PM    <DIR>          .
04/23/2025  07:05 PM    <DIR>          ..
07/16/2024  05:53 PM           235,008 sdrplay3_python.cp311-win_amd64.pyd
04/23/2025  07:05 PM               845 __init__.py
04/23/2025  07:05 PM    <DIR>          __pycache__
               3 File(s)        248,141 bytes
               3 Dir(s)  408,431,828,992 bytes free
```

In particular it has the ```__init__.py``` file that will be run when we import sdrplay3.

We can try to do the import from the radioconda prompt the same way we print sys.path:

```
(base) C:\Users\shaq\radioconda>python -c "import gnuradio; from gnuradio import sdrplay3"
Traceback (most recent call last):
  File "C:\Users\shaq\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 28, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\shaq\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 32, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'
```

So we run ```__init__.py``` which then tries to ```import .sdrplay3_python``` which fails.

We can prove the error comes from our copy of `Lib\site-packages\gnuradio\sdrplay3\__init__.py` and using an editor to add the following at line 23 ( right after 'import os' )

`import sys; print('\n'.join(sys.path))`

It's the same code we used from the command line to print sys.path.

Running the command again should print sys.path if we are indeed running the code in the `Lib\site-packages\gnuradio\sdrplay3` folder, and it does:

```
(base) C:\Users\shaq\radioconda>python -c "import gnuradio; from gnuradio import sdrplay3"

C:\Users\shaq\radioconda\python312.zip
C:\Users\shaq\radioconda\DLLs
C:\Users\shaq\radioconda\Lib
C:\Users\shaq\radioconda
C:\Users\shaq\radioconda\Lib\site-packages
C:\Users\shaq\radioconda\Lib\site-packages\win32
C:\Users\shaq\radioconda\Lib\site-packages\win32\lib
C:\Users\shaq\radioconda\Lib\site-packages\Pythonwin
Traceback (most recent call last):
  File "C:\Users\shaq\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 28, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\shaq\radioconda\Lib\site-packages\gnuradio\sdrplay3\__init__.py", line 32, in <module>
    from .sdrplay3_python import *
ModuleNotFoundError: No module named 'gnuradio.sdrplay3.sdrplay3_python'
```

We see the print of sys.path so it is indeed running that exact code in that exact folder.

So we know it isn't a problem of finding the right code, if it finds ```__init__.py``` it will find the ```.pyd``` file containing ```.sdrplay3_python``` code.

What I believe this means is that Python is trying to read the `sdrplay3_python.cp311-win_amd64.pyd` file in the `Lib\site-packages\gnuradio\sdrplay3` folder and it is failing.

What I believe the cause is that the version of Python that created the `.pyd` file we downloaded is incompatible with the version of Python used by radioconda.

That version is:

```
(base) C:\Users\shaq\radioconda>python --version
Python 3.12.9
```

Maybe someone can share which python built the `.pyd` file?

I tried adding code to `__init__.py` to find out more info based on https://stackoverflow.com/questions/5099931/figuring-out-what-really-went-wrong-when-an-importerror-is-raised but it didn't tell much if anything beyond what we knew already.

One thing it does remind us is there are other reasons why import fails, it's not just a missing file.

I suppose building gr_sdrplay3 from source as the instructions describe would prove or disprove the theory since the pyd file would be built by the radioconda python.  If the gr-sdrplay3 code could be built by gcc then we could do that under radioconda.  

So I don't think the mystery is solved but I do think some of the more probable causes are probably eliminated.

fventuri, 2025-04-24:
Thanks for looking into this issue with the `gr-sdrplay3` module under radioconda on Windows.
I think the `cp311` in the Python DLL filename stands for CPython 3.11, which would be the version of Python I had on radioconda when I built it.

If you have Visual Studio 2022 on your PC (the Community Edition is free), you should be able to rebuild it with your more recent version of radioconda and Python following the instructions here: https://github.com/fventuri/gr-sdrplay3/blob/main/WINDOWS.md#step-3---optional-build-gr-sdrplay3-from-source

Franco

n1ai, 2025-04-24:
Yes, I was reading that as well as https://github.com/fventuri/gr-sdrplay3/blob/main/.conda/README.md .... 

It seems you know how to get this built under conda.  I'm wondering if you've asked Ryan about getting this added to radioconda?  I have worked with him to get some code built under radioconda and he's super easy to work with.  If we could do this, then we could be sure the gr-sdrplay2 code would be built with the same tools as radioconda is.

JLRMSousa, 2025-04-25:
Franco,
I have been going through some of the debug pages I have found. It is a bit confusing, because some are for Linux only. some commands work in the win11 powershell and some work on the radioconda Prompt.
I confirmed the existence of 
C:\Users\me\radioconda\Library\bin\sdrplay_api.dll
But this path is not part of the python path as listed in my previous message, so I copied sdrplay_api.dll to C:\Users\me\radioconda\Library, which is listed there. This had no effect. 

I also searched the entire C: drive for copies of python.exe and got this:
(base) C:\Users\me> where python
C:\Users\me\radioconda\python.exe
C:\Users\me\AppData\Local\Microsoft\WindowsApps\python.exe

Does this mean I have two installations of Python? 
The second line of these is dated 3/19/2025, the first line is dated 3/4/2025.
My radioconda directory is dated 4/18/25

Regards,
-Joe

JLRMSousa, 2025-04-25:
Following suggestions from @n1ai , I finally got success!

@n1ai thought that the problem might be a version incompatibility between radioconda and gnuradio. 
I uninstalled radioconda-2025.03.14-Windows-x86_64.exe 
and installed the older radioconda-2024.05.29-Windows-x86_64.exe that is contemporary to 
gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2
After installing gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2, it all finally worked.
Listening to the "C:\Users\me\radioconda\Library\share\gr-sdrplay3\examples\FM-Receiver-SDRplay.grc" on the main tuner of my RSPduo as I write this.

It is established that in my Windows11 PC, this radioconda version works with this gnuradio:
 radioconda-2024.05.29-Windows-x86_64.exe 
 gnuradio-sdrplay3-3.11.0.7-py311h702a0ab_0.tar.bz2

Thank you all for the extraordinary effort and help!
-Joe

fventuri, 2025-04-25:
Joe,
I just read your message now after I finished writing mine.

I am really glad that you with the help of @n1ai were able to figure it out!

Franco

n1ai, 2025-04-25:
> I also searched the entire C: drive for copies of python.exe and got this:
(base) C:\Users\me> where python
C:\Users\me\radioconda\python.exe
C:\Users\me\AppData\Local\Microsoft\WindowsApps\python.exe
> 
> Does this mean I have two installations of Python?

It is normal to have two copies when using conda/radioconda.

As soon as you do `conda activate` or run in the radioconda command line prompt window, the runtime has been changed so radioconda's version of Python is the one in use. 

It's for the reason we learned the hard way: You really have to make sure all the various binary files like .pyd files (which are just .dll files with a different suffix) are built in the same environment that the Python interpreter was built in.  

The conda/radioconda environment definitely can't assume that the user has installed a compatible Python runtime.  

One nice thing about the conda world is that it does provide this unique space (by using something close to what Python refers to as a venv).  On top of that, it knows all the binaries in its space are compatible with all the other components, because its installer has very detailed information about the dependencies between all the objects in its space, and makes it very difficult for a user to force an install.

It's a far better package manager (IMO) than what is provided by pypi, because while it has some dependency info it lets users force an install which is pretty much guaranteed to blow up eventually.

At the risk of being repetitive, integrating gr-sdrplay2 into radioconda would mean all these things stay in sync with each other, but that's totally up to you.

fventuri, 2025-04-25:
@n1ai  - I had a discussion about that with Ryan Volz in the GNU Radio discussion mailing list a few years ago: https://lists.gnu.org/archive/html/discuss-gnuradio/2022-03/msg00146.html

Basically, because of the proprietary license of the SDRPlay API, I am afraid it is not possible to use conda-forge for this GNU Radio OOT module like all the other ones.

Another unrelated issue is that the most current code for `gr-sdrplay3` does build with radioconda on Windows, but when I (or anyone else) tries to run it, it throws this error `The specified procedure could not be found`, that I haven't figured out yet (but it builds and runs without problems on Linux).
I opened an issue about it in the radioconda repository a few months ago (https://github.com/ryanvolz/radioconda/issues/116), but I don't think Ryan has had time to look into it yet.

Franco

PS: I noticed that you wrote `gr-sdrplay2` (instead of `gr-sdrplay3`) - do you perhaps refer to a different GNU Radio OOT module that uses the old SDRplay API version 2.X?

n1ai, 2025-04-25:
Ok, I imagined there could be some sort of issue with the binaries of the sdrplay API, and of course shipping with a known problem like that should be avoided.

Indeed I wrote gr-sdrplay2 where I should have written gr-sdrplay3, thanks for the correction!

fventuri, 2025-05-09:
Last night I was able to build a conda package for 'gr-sdrplay3' version 3.11.0.8 with the latest radioconda (2025.03.14).
I just published a new release with that build: https://github.com/fventuri/gr-sdrplay3/releases/tag/v3.11.0.8

I also updated the instructions for Windows (https://github.com/fventuri/gr-sdrplay3/blob/main/WINDOWS.md) because I found out that `conda update --all` was causing problems with GNU Radio. One or more of those update packages is causing GNU Radio to throw this error:  `DLL load failed while importing ...: The specified procedure could not be found`.
I added that info to the Windows instructions in case someone else has the same problem.

Give it a try with this latest version, and let me know how it goes.

Franco