fventuri/gr-sdrplay3 #58: Merging changes from `message-passing` branch into `main`

jcfitzpatrick12, 2025-04-21:
Hi Franco,

Thanks again for your significant help enabling a frequency sweep mode in [`spectre`](https://github.com/jcfitzpatrick12/spectre) for RSP devices. Last year, I managed to capture an [enormous radio burst](https://github.com/jcfitzpatrick12/spectre/blob/main/docs/gallery/solar_radio.png) on an RSP1A over a 40MHz frequency range using the code on your `message-passing` branch. 

Currently, during the docker build for the `spectre-server`, I am building the `gr-sdrplay3` OOT module by checking out the message passing branch directly (see  [this line](https://github.com/jcfitzpatrick12/spectre/blob/main/backend/Dockerfile#L50)). To make the build more robust, it would be useful if I could refer to a tagged version on the master branch.

I understand it's been a while since any work has been done on the `message-passing` branch so this might be non-trivial. In the same breath, I am wanting to learn more about C++ and the SDRplay API, so if I can do anything to help do let me know :)

Jimmy

fventuri, 2025-04-22:
Jimmy,
I am really happy to hear that the gr-sdrplay OOT module help you make some really interesting radioastronomy work using `spectre`!

There's one more thing I want to do before merging that code in the `main` branch, which is switching the way messages are sent to a specific message port for the center frequency (the way it is now) to instead having a general purpose `command` port where center frequency is one of the several settings that can be changed (gains are the other main setting I imagine people would be interested in).
The latter is the way the UHD GNU Radio blocks work (if I remember correctly), so I thought it would be nice to have some level of consistency between the two.

I am 'almost done' (famous last words) with another project these days.
After that, I plan to start looking into this change, and then I'll merge the `message-passing` into `main`.

Franco

jcfitzpatrick12, 2025-04-23:
Sounds good to me, and good luck with your current project !

Coming to think of it, to keep my builds stable if you're developing on that branch I can fork it in its current state too. 

In any case, consistency with the UHD GNU Radio blocks would be nifty. For your interest, I have been able to frequency swept spectrograms with my b200mini in the same way via message passing:

 [image]

fventuri, 2025-05-10:
Jimmy,
I just sync'd the `message-passing` branch with the changes I made recently to the `main` branch.

I also changed the message port to `command`, and it now expects a dict (keys and values).
To set the center frequency you have to send a message to that `command` port with key set to `freq` and as the value the center frequency you want to switch to. I read the USRP code and I think this is how it works too.
To make sure it can handle multiple settings, I also implemented the gains; their keys are `if_gain`, `rf_gain`, and `lna_state`.

Once you confirm that everything works with the frequency sweep in `spectre`, then I plan to add a few other settings to that message port (which should be straightforward at this point), and then merge all these changes into `main`.
That will become the next release (3.11.0.9) of this `gr-sdrplay3` module.

I really appreciate your help with testing these changes,
Franco

jcfitzpatrick12, 2025-05-11:
Hi @fventuri, sorry for the delayed response. All this sounds great, and I can test these changes over the next week or so. I'm currently refactoring the Dockerfile, which by-the-by will make it clearer and simpler to add/modify OOT modules to the final runtime image. Right now, it definitely includes some "junk" - a product of some ham-fisted build stages to get things working. It's been a challenge wrangling the GNU Radio dependencies together (hardware drivers, OOT modules, etc.) - but god bless multi-stage builds ! 

Only loosely related, I've recently been tinkering with the SDRplay API directly for the first time—the closest I've worked with hardware yet, though still a long way from the bare metal ! It's an interesting set-up with the SDRplay API service running in the background.

fventuri, 2025-05-11:
Jimmy,
no rush about testing these changes. As far as I know, you are the only one using message ports with this `gr-sdrplay3` OOT module, so take your time.

As per exploring the SDRplay API and calling it directly, you'll find it is not that hard, and I really hope it will help you with the frequency sweeep in `spectre`.
If you decide to learn more about it, you may want to look at a couple of repositories I have here: `single-tuner-experiments` (https://github.com/fventuri/single-tuner-experiments) and `dual-tuner-experiments`  (https://github.com/fventuri/dual-tuner-experiments), and also the the new `dual-tuner-recorder` (https://github.com/fventuri/dual-tuner-recorder).
This last one is the project I mentioned to you the other day that I was working on.

Franco

jcfitzpatrick12, 2025-05-23:
@fventuri 

Thanks for your patience - I've updated things on my end to accommodate your new changes and it's working great ! Attached image is a swept spectrogram from an RSP1A over ~15MHz in the FM band at a 5MHz sample rate. I'm doing a reasonably quick sweep here (retuning every 50ms), resulting in the artefact at ~16:07:55 where the samples get mixed up. 

[image]

Lastly, I appreciate the repo links, I've been taking a look they've been really helpful :)

Jimmy

jcfitzpatrick12, 2025-06-02:
@fventuri  great stuff - let me give it a go today 📡